HASwitchPlate/openHASP #283: Add FTP Service

fvanroie, 2022-01-17:
**Is your feature request related to a problem? Please describe.**

Uploading multiple files to a new plate via the File Editor can be tedious. We have already added the *uncompressed zip* option to facilitate this a little bit. The upload of zip file still requires unpacking of the archive using the `unzip` command.

I discovered a new repository that implements a [SimpleFTPServer ](https://github.com/xreef/SimpleFTPServer) on ESP32 and Arduino. This seems like a good fit to add to openHASP. It makes it a lot easier to upload and download files from the plate.

**Describe the solution you'd like**

Add [SimpleFTPServer ](https://github.com/xreef/SimpleFTPServer) as an openHASP service, like `telnet` or `http`.

**Describe alternatives you've considered**

`unzip` command is still not very user friendly. FTP is a well known file transfer protocol with multi-platform client support.

**Additional context**

Some issues still need to be resolved:
- [ ] Arduino v2 uses `LittleFS` instead of `LITTLEFS`
- [ ] ⚠️ Reading of `/config.json` poses a security issue
- [ ] The service can not be stopped (so `service stop ftp` won't work)
- [ ] At this time the service is always started if `HASP_USE_FTP=1` and `HASP_START_FTP=1` is defined

kquinsland, 2022-01-19:
Is ftp really easier than `curl` with the http server that's already in there?

```shell
❯ curl http://roomPlateIpv4Here/edit --form name=data --form data=@pages.jsonl
```

This will upload the `pages.jsonl` file from `$PWD` to the device. You can wrap this in a simple `for ... in ...` loop:

```shell
❯ for i in *.jsonl; do
echo uploding $i
curl http://roomPlateIpv4Here/edit --form name=data --form data=@$i
done
```

or similar

fvanroie, 2022-01-19:
It depends, if you're an admin or developer that's familiar with `curl`, by all means... I hadn't even thought of using that.😃 
But for a regular user who's accustomed to using a GUI application, I think FTP will be easier.

kquinsland, 2022-01-19:
Fair enough. Browsers don't support FTP anymore so I didn't see a huge difference between installing curl and installing another FTP program :/

fvanroie, 2022-01-24:
I use SFTP fairly often, so already have a client for that. It seemed quite useful and suitable for the project.
SFTP/SSH is a bit much for the ESP32, but on a LAN it could be OK.

nagyrobi, 2022-01-29:
FTP only works if the client and the plate are on the same network. Impossible to connect through FTP from a different subnet.

Filezilla default settings (max connections 1):
```
Status:    Connection established, waiting for welcome message...
Status:    Plain FTP is insecure. Please switch to FTP over TLS.
Status:    Server does not support non-ASCII characters.
Status:    Logged in
Status:    Retrieving directory listing...
Command:    PWD
Response:    257 "/" is your current directory
Command:    TYPE I
Response:    200 TYPE is now 8-bit binary
Command:    PORT xxx,xxx,xxx,xxx,xxx,xxx
Response:    200 PORT command successful
Command:    MLSD
Response:    425 No data connection
Error:    Failed to retrieve directory listing
```
Passive mode:
```
Status:    Connection established, waiting for welcome message...
Status:    Plain FTP is insecure. Please switch to FTP over TLS.
Status:    Server does not support non-ASCII characters.
Status:    Logged in
Status:    Retrieving directory listing...
Command:    PWD
Response:    257 "/" is your current directory
Command:    TYPE I
Response:    200 TYPE is now 8-bit binary
Command:    PASV
Response:    227 Entering Passive Mode (0,0,0,0,195,89)
Command:    MLSD
Error:    The data connection could not be established: ECONNREFUSED - Connection refused by server
Response:    425 No data connection
Error:    Failed to retrieve directory listing
```
If this is implemented deliberately as a kind of security measure, it should rather be disregarded. In secure environments IoT devices should be placed in separate subnets to be able to isolate their traffic, and management has to be possible to be done from user networks through normal routing policies - and FTP by design supports that - I think there's some extra constraint here which is contraproductive imho.

nagyrobi, 2022-01-29:
Looking at the code of the FTP library, it uses a static port for PASV which is 50009. The error above complains that the client can't connect to this port, `Connection refused` by the plate.

nagyrobi, 2022-01-29:
https://github.com/xreef/SimpleFTPServer/issues/8

kquinsland, 2022-01-29:
> FTP now only works if the client and the plate are on the same network.

This this really the case? If yes, how can I disable/turn this off?  I deploy my openHasp devices to a subnet that is different from the one where my workstation resides.  I would prefer not writing a SNAT rule on my router to trick the openHasp devices into thinking that the FTP traffic from my workstation is coming from another device on the openHasp subnet...

nagyrobi, 2022-01-29:
Read the issue above. The SimpleFTPServer has a bug, it reports 0.0.0.0 as the PASV IP address...

nagyrobi, 2022-01-31:
Works great. It allows for a simple and effective way to do backups off all config. Excellent.

AlbrechtL, 2023-01-29:
Thanks for this nice feature! With this was able to upload all my files at once.

Just one hint: There is a lack of documentation which FTP username and password needs to be used. Actually, on the openhasp homepage there is no documentation about FTP at all.

Anyway, I found the answer in the code here https://github.com/HASwitchPlate/openHASP/pull/284/files#diff-b07f16599331cf75146b1a06431eaf8a01488bb416807415c703958cf742782dR113. The HTTP username and password is used. So I needed to set is first. In Filezilla I activated the passiv mode and limit the numbers of connection to 1.

Maybe this information helps others to use the great FTP access.

fvanroie, 2023-01-31:
@AlbrechtL Thanks!

The link with the HTTP credentials was just a test to see if the library was working.
Now that this feature is stable, I've updated the library and added specific FTP configuration options in the Web UI.
We are working on the 0.7.0-dev docs, and this service will be documented there eventually.