HASwitchPlate/openHASP #285: Syslog messages are not properly formatted

fvanroie, 2022-01-24:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
As reported by @geiseri in issue: *Allow ANSI codes to be disabled for syslog #261* the syslog messages are garbled.

### To Reproduce
Configure syslog and watch the output below.

### Expected behavior
Properly formed IETF (RFC 5424) / BSD (RFC 3164) syslog messages to be sent.

### Screenshots or video
[image]

fvanroie, 2022-01-25:
BSD-syslog or legacy-syslog messages look OK now. I don't have setup to test IETF-syslog messages at this moment.
Feel free to test it 😃

geiseri, 2022-02-02:
When testing with rsyslog-ng I split the logs by the hostname in the logline.  It seems to be okay like 99% of the time, but then randomly I get hostnames with stuff like `xt` and sometimes a random number.  I looked at the contents of one with the hostname of `xt` and it had the following:
```
2022-02-02T23:42:24.673151+00:00 xt = 
2022-02-02T23:42:24.703462+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0
2022-02-02T23:42:24.721584+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = 
2022-02-02T23:42:24.750760+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0
2022-02-02T23:42:24.772555+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0[18:42:24.653][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b1.enabled = False
2022-02-02T23:42:24.791641+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0[18:42:24.653][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b1.enabled = False[18:42:24.674][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b1.color =
2022-02-02T23:42:24.806179+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0[18:42:24.653][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b1.enabled = False[18:42:24.674][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b1.color = [18:42:24.685][65524/114340 42][35220/35272  1] MQTT PUB: p3b1 => {"color":"#ff0000","r":255,"g":0,"b":0}
2022-02-02T23:42:24.827268+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0[18:42:24.653][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b1.enabled = False[18:42:24.674][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b1.color = [18:42:24.685][65524/114340 42][35220/35272  1] MQTT PUB: p3b1 => {"color":"#ff0000","r":255,"g":0,"b":0}[18:42:24.707][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b3.enabled = False
2022-02-02T23:42:24.847796+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0[18:42:24.653][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b1.enabled = False[18:42:24.674][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b1.color = [18:42:24.685][65524/114340 42][35220/35272  1] MQTT PUB: p3b1 => {"color":"#ff0000","r":255,"g":0,"b":0}[18:42:24.707][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b3.enabled = False[18:42:24.727][65524/114004 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b3.val = 0
2022-02-02T23:42:24.871363+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0[18:42:24.653][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b1.enabled = False[18:42:24.674][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b1.color = [18:42:24.685][65524/114340 42][35220/35272  1] MQTT PUB: p3b1 => {"color":"#ff0000","r":255,"g":0,"b":0}[18:42:24.707][65524/114340 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b3.enabled = False[18:42:24.727][65524/114004 42][35220/35272  1] MQTT RCV: hasp/alarm_clock/command/p3b3.val = 0[18:42:24.751][65524/112508 41][35220/35272  1] HTTP: Sent / page to 10.0.5.205
2022-02-02T23:42:24.909428+00:00 xt = [18:42:24.584][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.val = 0[18:42:24.604][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p1b12.text = [18:42:24.633][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/command/p3b2.val = 0[18:42:24.653][65524/114340 42][34812/35104  1] MQTT RCV: hasp/alarm_clock/co [... cut ...] 65524/114104 42][35220/35744  2] MQTT RCV: hasp/alarm_clock/command/p2b66.val = 23.1376953125[18:49:45.734][65524/114104 42][35220/35664  2] MQTT RCV: hasp/alarm_clock/command/p2b67.val = 31.71484375[18:49:45.954][65524/114104 42][35220/35744  2] MQTT RCV: hasp/alarm_clock/command/p2b68.val = 51.6923828125
```
It's just that in some cases the first part of the line is not the host IP address, it's some random truncated text.  So I wonder if there is an underlying issue here with the buffer size.

fvanroie, 2022-03-02:
There is no real buffer reserved for the syslog messages except for the UDP packet size.
So maybe these messages are overflowing the UDP buffer....

geiseri, 2022-03-03:
I will see if I can use wireshark to find something.  It smells like somehow the buffer isn't clearing and its concatenating, but it could be like you said and the messages are overflowing a buffer on the rsyslog end.

fvanroie, 2022-04-12:
Since this is working 99% of the time I will consider this issue as resolved.
If you can provide some hints to when the 1% of the messages are not properly formatted, feel free to re-open it.

closed 2022-04-12 by fvanroie