HASwitchPlate/openHASP #295: Button's `text_color` in ON state is ignored

nagyrobi, 2022-01-28:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

After a toggle button is pressed, setting a new `text_color` property does not change the color of the text on the button.

### To Reproduce
Plate design (`plate_teszt`, Lanbon L8, toggles acting on the first two relays):
```json
{"page":1,"id":2,"obj":"btn","x":125,"y":40,"w":105,"h":105,"toggle":true,"text":"\uE335","text_font":32,"mode":"break","align":1,"groupid":1,"value_str":"Button A","value_color":"white","value_ofs_y":"30"}
{"page":1,"id":3,"obj":"btn","x":10,"y":40,"w":105,"h":105,"toggle":true,"text":"\uE335","text_font":32,"mode":"break","align":1,"groupid":2,"value_str":"Button B","value_color":"white","value_ofs_y":"30"}
```
CC config:
```yaml
  plate_teszt:
    objects:
      - obj: "p1b2"
        properties:
          "text_color": "{{ '#FFFF00' if is_state('light.plate_teszt_light_12', 'on') else '#FFFFFF' }}"

      - obj: "p1b3"
        properties:
          "text_color": "{{ '#FFFF00' if is_state('light.plate_teszt_light_14', 'on') else '#FFFFFF' }}"
```
While turning OFF the button, the yellow color flashes up momentarily. It looks like the property is actually correctly set, but presumably some other styling item (from the theme maybe?) overrides this, and clearing of this override happens faster than the CC sends the command.

Reproducible with all the themes on the plate.

### Expected behavior
When buttons are actively depressed, the lightbulb icon color should be yellow. When released, should be white.

### Screenshots or video

fvanroie, 2022-01-29:
Also add the `text_color3` property.

nagyrobi, 2022-01-29:
I added `"text_color3":"#FFFF00"` to the jsonl of 3 buttons in the design file, saved it and rebooted the plate. Also, I removed from CC configuration the state-based colorisation.

`text_color3` exhibits a very strange behavior:
- only activates when the button is touched physically. If the value is set though MQTT, it doesn't apply the color.
- if there are multiple toggle buttons on the page, only one of them activates the color. For example if I touch the first button, it will become yellow when toggled down. If I touch the second one, that one will become yellow instead, the first one will be white again. 

It behaves like a way to know which was the last button pressed on the page, which is not what the goal is. The goal is to have the text on the button in a different color, when the button is toggled down.

nagyrobi, 2022-01-30:
It's the same issue as: https://github.com/HASwitchPlate/openHASP/discussions/108#discussioncomment-2002207

fvanroie, 2022-01-31:
Set the color in pairs:
```c
p1b2.text_color white
p1b2.text_color2 white

p1b2.text_color1 yellow
p1b2.text_color3 yellow
```

nagyrobi, 2022-01-31:
It still conflicts when two or more buttons are on the same page. The second button even if it's toggled on, changes back to white when toggling off the first one.

```yaml
      - obj: "p1b2"
        properties:
          "text_color": '{{ "white" if is_state("light.plate_teszt_light_12", "off") }}'
          "text_color2": '{{ "white" if is_state("light.plate_teszt_light_12", "off") }}'
          "text_color1": '{{ "yellow" if is_state("light.plate_teszt_light_12", "on") }}'
          "text_color3": '{{ "yellow" if is_state("light.plate_teszt_light_12", "on") }}'

      - obj: "p1b3"
        properties:
          "text_color": '{{ "white" if is_state("light.plate_teszt_light_14", "off") }}'
          "text_color2": '{{ "white" if is_state("light.plate_teszt_light_14", "off") }}'
          "text_color1": '{{ "yellow" if is_state("light.plate_teszt_light_14", "on") }}'
          "text_color3": '{{ "yellow" if is_state("light.plate_teszt_light_14", "on") }}'
```

nagyrobi, 2022-01-31:
Ah!
Seems to work without any dynamic property modification, just with this in jsonl:
`"text_color":"white","text_color2":"white","text_color1":"yellow","text_color3":"yellow"`

closed 2022-01-31 by nagyrobi

nate922, 2022-04-13:
@fvanroie this issue also occurs with bg_color, can we please apply the same fix?

Thank you.

fvanroie, 2022-04-14:
@nate922 Yes it shoud. What version of openHASP are you running?

nate922, 2022-04-14:
@fvanroie 0.6.2, I'm assuming that 0.6.3 beta has this fixed? 

Does the beta have the Lanbon L8 SPI_FREQUENCY fix? I had to recompile 0.6.2

Thanks so much.

fvanroie, 2022-04-14:
> I'm assuming that 0.6.3 beta has this fixed?

You'll need to use `0.6.3-dev` because it has the improved support for state & parts. It's still quite new and not very well tested or documented... so your mileage may vary. Let me know if you find any issues.

> Does the beta have the Lanbon L8 SPI_FREQUENCY fix? I had to recompile 0.6.2

Yes, in `0.6.3-dev` the default Lanbon L8 `SPI_FREQUENCY` is set to 60MHz instead of 80Mhz in `0.6.2`.

nate922, 2022-04-14:
Thanks @fvanroie