HASwitchPlate/openHASP #299: Multiple dropdown list in tabview has irratic behaviour when using open command on object

htvekov, 2022-01-30:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

When shifting between two dropdown lists, connected via a shared tabview and using forced open on the dropdownlist chosen, behaviour is somewhat irratic. It seems that dropdown list object sort of inherits the 'mother' parent tabviews value and prevents `"event":"changed"` to be performed for precisely that value ?

At least it  seems like it ,as first tab (tabview `"val":0`) won't send event changes for its 'own' `"val":0` on the list. And for the second dropdown list its `"val":1` that behaves like this.

Issue arises **only** when i use the force open command (which I need)

If you shift back and forth between the two tabs and everytime select one item on the list, you can see that no `"event:changed"` is send for **"val":0** for `"id":10` and `"val":1` for `"id":11`  Only up/down events ??

If the `"open:0"` is removed from last service call's jsonl command (default behaviour), then all items on both dropdown lists will send `"event:changed"` mqtt state updates.

### To Reproduce
Below jsonl for plate and automations.yaml for HA to reproduce issue.

```json
{"page":4,"comment":"---------- Page 4 ----------"}
{"page":4,"id":95,"obj":"tabview","btn_pos":1,"x":1,"y":50,"h":255,"hidden":0}
{"page":4,"id":96,"obj":"tab","parentid":95,"text":"TAB 1 - id:10"}
{"page":4,"id":97,"obj":"tab","parentid":95,"text":"TAB 2 - id:11"}
{"page":4,"id":10,"parentid":96,"obj":"dropdown","open":1,"x":010,"y":10,"w":105,"h":30,"direction":0,"options":"id:10 val:0\nid:10 val:1\nid:10 val:2\nid:10 val:3"}
{"page":4,"id":11,"parentid":97,"obj":"dropdown","open":1,"x":120,"y":10,"w":105,"h":30,"direction":0,"options":"id:11 val:0\nid:11 val:1\nid:11 val:2\nid:11 val:3"}
```

automation.yaml
Default openhasp mqtt topic used

```yaml
- id: test_tabview_multiple_dropdownlist_issue
  alias: test_tabview_multiple_dropdownlist_issue
  mode: single
  trigger:
    - platform: mqtt
      topic: hasp/+/state/p4b95
  condition:
  - condition: template
    value_template: "{{ trigger.payload_json.event == 'changed' }}"
  action:
  - service: mqtt.publish
    data:
      topic: >-
        hasp/{{trigger.topic.split('/')[1]}}/command
      payload: >-
        {% if trigger.payload_json.val == 1 %}
          {"page":4,"id":10,"close":0}
        {% elif trigger.payload_json.val == 0 %}
          {"page":4,"id":11,"close":0}
        {% else %}
        {% endif %}
  - service: mqtt.publish
    data:
      topic: >-
        hasp/{{trigger.topic.split('/')[1]}}/command
      payload: >-
        {% if trigger.payload_json.val == 1 %}
          {"page":4,"id":11,"open":0}
        {% elif trigger.payload_json.val == 0 %}
          {"page":4,"id":10,"open":0}
        {% else %}
        {% endif %}
```

### Expected behavior

Expected behaviour would be that open command for dropdown list object would be identical with a physical touch on display to fold out list.

### Screenshots or video

fvanroie, 2022-02-01:
Event handlers prevent the last value from being sent repeatedly. Normally the value is reset on the next `down` event.
But when sending a `messagebox.open` command there is no such reset of the previous value...

I pushed a *possible* fix for this issue which forces `messagebox.open` to always reset the last value.
This should fix the issue and always let the next touch be recognized.

Please test if this fixes your issue and I hope I did not break anything else in the process 🤣

htvekov, 2022-02-01:
Just did a quick test and I think you've (once again) nailed it ! 🤗🤗👍
Haven't experienced any other 'sideeffects' with this release.

F*ucking awesome !!! 😎

Rgds.
Henning

fvanroie, 2022-02-02:
Great. Thanks for reporting and testing!

closed 2022-02-02 by fvanroie