HASwitchPlate/openHASP #300: Firmware Upgrade From URL

nagyrobi, 2022-02-02:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
In the firmware upgrade web interface, updatding from URL is not working. Updating using the browser's upload form works fine.

### To Reproduce
Put the OTA binary to a web server accessible via plain HTTP. Eg:
http://building.rig.local/code/openhasp/openHASP/build_output/firmware/lanbon_l8_ota_v0.6.3-dev.bin
Verify that downloading of the file from the URL is working fine using a PC.

Open a Telnet session to the plate.
Copy the URL in openHASP's _From URL_ form field and press _Update from URL_ button.
The web interface will show
```
Updating ESP firmware from: http://building.rig.local/code/openhasp/openHASP/build_output/firmware/lanbon_l8_ota_v0.6.3-dev.bin
```
After a while the page will return to root.

Telnet will show:
```
[10:44:41.533][65524/89868 27][19536/19944  3] HTTP: Sent /firmware page to 192.x x x
[10:44:50.278][65524/89660 26][19336/19940  4] HTTP: Sent /espfirmware page to 192.x x x
[10:44:50.312][65524/88212 25][19336/19940  4] HTTP: Updating ESP firmware from:
[10:45:05.515][65524/89868 27][19336/19940  4] HTTP: Sent / page to 192.x x x
```

### Expected behavior
It should update from URL.
Also it should "remember" the URL between the updates so that just replacing the binary on the server should be enough before pressing the _Update from URL_ button (no need to paste it every time).

### Screenshots or video
N/A

fvanroie, 2022-05-05:
Update from URL is enabled again in 0.6.3-dev.

The 0.6.2 release contains a bug that requires you to power-cycle the plate instead of the "normal" reboot. 0.6.2 actually doesn't enable the new firmware until after a power-cycle 😞 

This should be fixed now in 0.6.3-dev, so updating to a newer 0.6.3 code-base can be done via Web upload, Web URL or Arduino OTA without power-cycle.

nagyrobi, 2022-05-05:
Tested, works.

fvanroie, 2022-05-06:
It's a fixed time-out that goes back to the home page. The download, update and reboot process takes a variable time, so it's hard to guesstimate how long it will take. To prevent this, you would need to do active javascript polling...

nagyrobi, 2022-05-06:
Just add +3 seconds...

fvanroie, 2022-05-06:
done

fvanroie, 2022-05-07:
Last URL is now saved. There is now also the option to follow redirects or not, this allows direct downloads from e.g. Github urls that redirect to a CDN for downloading of the actual file.

fvanroie, 2022-05-12:
The feature is now part of [0.6.3-rc1](https://github.com/HASwitchPlate/openHASP/discussions/334)

closed 2022-05-12 by fvanroie

nagyrobi, 2022-09-14:
In 0.6.4-dev 7f4e079 it's broken, OTA from URL is impossible to complete. Bar goes to the end on the screen but the plate doesn't reboot, instead the web interface jumps back to main menu. Rebooting from there doesn't result in booting from the new firmware image.