HASwitchPlate/openHASP #303: Support offset coordinates for objects

jbraceg, 2022-02-09:
It would be great if objects let you specify their x/y/w/h as offsets from the another object. Very often I find myself wanting to create several objects aligned together, but then moving them all requires quite fiddly arithmetic on the values. Maybe you could support a syntax of "id+v" or "id-v" that indicates a difference from the specified object id. Then I could do this:

{"page": 1,"id": 5,"obj": "label", "x": 90,"y": 295,"w": 200,"h": 25,"text": "blah","text_font": 16,"align": "left"}
{"page": 1,"id": 6,"obj": "label", "x": "5+0","y": "5+30","w": "5+0","h": "5+0","text": "blah","text_font": 16,"align": "left"}
{"page": 1,"id": 7,"obj": "label", "x": "6+0","y": "6+30","w": "6+0","h": "6+0","text": "blah","text_font": 16,"align": "left"}

And I could move all three components around by simply changing the values of object 5.

dgomes, 2022-02-09:
we fully rely in HA template engine...

jbraceg, 2022-02-09:
I'm not referring to values supplied by HA, I'm referring to the hasp design pages. I believe it would require changes in hasp_attribute.cpp etc to handle the new values.

Actually, thinking about this a little more, it could also be done by a pre-processor generating the jsonl file. That's probably the less-intrusive approach, I might be able to create that myself.

dgomes, 2022-02-09:
sorry :) I was actually thinking you wanted to generate the jsonl in the CC

fvanroie, 2022-02-09:
Try adding a parent `obj` object and set the `parentid` property for its children :
| Property | Value        | Required | Default | Description
|:---------|:------------:|:--------:|:-------:|:----
| parentid | 0..255 | no       | 0       | Set the object to be the child of another object.<br>`x` and `y` will be relative to the parent object. _Property inheritance can affect the appearance of the of the children (more info in styling).

something like this:

```json
{"page": 1,"id":10,"obj": "obj", "x":5, "y":35, "w":230, "h":250, "click":0}
{"page": 1,"id": 5,"obj": "label", "parentid":10, "x": 90, "y":295,"w": 200, "h": 25,"text": "blah","text_font": 16,"align": "left"}
{"page": 1,"id": 6,"obj": "label", "parentid":10, "x":  0, "y": 30,"w": 200, "h": 25,"text": "blah","text_font": 16,"align": "left"}
{"page": 1,"id": 7,"obj": "label", "parentid":10, "x":  0, "y": 30,"w": 200, "h": 25,"text": "blah","text_font": 16,"align": "left"}
```
Then you can move the parent object around and the children will follow it.
By styling the base object you can make it look like a panel or have it transparent...

> Maybe you could support a syntax of "id+v" or "id-v" that indicates a difference from the specified object id.

That syntax doesn't look intuitive since `id` refers to an object and x/y/w/h are in pixels.
It adds further overhead on the MCU to parse these values, while we want them to render as fast as possible.

jbraceg, 2022-02-09:
Amazing! I don't know how I missed this. Thanks.

closed 2022-02-09 by jbraceg