HASwitchPlate/openHASP #304: MQTT Server Domain gets truncated when set via web interface

fake-name, 2022-02-14:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [ ] Update openHASP to the latest version
- [ ] Reproduce the issue and describe all steps

### Describe the bug

I have my mqtt server on a sub-domain of a domain I own.

When I set it, the trailing TLD get's truncated, which breaks it.

### To Reproduce

 - Access the HASP web-interface
 - Attempt to set mqtt broker address to `mqtter.fake-url.com`
 - mqtt broker address gets set to `mqtter.fake-url.` (yes, with trailing dot).
 - mqtt therefore doesn't work.

### Expected behavior

I'd expect setting the mqtt server to set the mqtt server.

### Screenshots or video

fake-name, 2022-02-14:
Ok, this seems to be a dupe of https://github.com/HASwitchPlate/openHASP/issues/150, but I literally just installed from https://github.com/HASwitchPlate/openHASP/releases/tag/v0.6.2 (which was released 6 months after that issue). 

Either the problem occurs in multiple places, or something didn't get merged?

fake-name, 2022-02-14:
Setting the domain via serial (`mqtthost <xxxx>`) also results in the truncation.

fake-name, 2022-02-14:
Pulling the source from the 0.6.2 release page:

```
char mqttServer[16]   = MQTT_HOST;
char mqttUser[23]     = MQTT_USER;
char mqttPassword[32] = MQTT_PASSW;
// char mqttNodeName[16]  = MQTT_NODENAME;
char mqttGroupName[16] = MQTT_GROUPNAME;
uint16_t mqttPort      = MQTT_PORT;
PubSubClient mqttClient(mqttNetworkClient);
```

So somehow #150 didn't make it into 0.6.2.

What happened?

fake-name, 2022-02-14:
Ok, grabbing a current release from CI (https://github.com/HASwitchPlate/openHASP/actions/runs/1834150765) fixes the issue.

Maybe cut a new release?

fvanroie, 2022-02-14:
This will be fixed (again) in the next release.

closed 2022-02-14 by fvanroie