HASwitchPlate/openHASP #307: Development branch compilation error

gheesung, 2022-03-02:
### Perform all steps below and tick them with [x]
- [X ] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [X ] Update openHASP to the latest version
- [X ] Reproduce the issue and describe all steps

### Describe the bug
Development branch compilation error 
At esp_merge_bin.py line 36
```
bootloader = "{}tools{}sdk{}esp32{}bin{}bootloader_dio_40m.bin".format(FRAMEWORK_DIR, os.path.sep, os.path.sep, os.path.sep, os.path.sep, os.path.sep)
```

"esp32" is not supposed to be in the path.

### Expected behavior
When the line is changed to below, the project can be built successfully.
```
 bootloader = "{}tools{}sdk{}bin{}bootloader_dio_40m.bin".format(FRAMEWORK_DIR, os.path.sep, os.path.sep, os.path.sep, os.path.sep, os.path.sep)
```

fvanroie, 2022-03-02:
@gheesung Thank you for reporting this issue. The path of the bootloader seems to vary depending on the ESP32 / ESP32S2 chip.

So I updated `esp_merge_bin.py` to fallback to the alternate path if the bootloader is not found at the first location.
This should fix the issue you are having, while also still work on the ESP32-S2 builds.

Please test it and let me know if the issue is resolved. Thanks again!

gheesung, 2022-03-03:
Thanks. It is building correctly now.

closed 2022-03-03 by gheesung