HASwitchPlate/openHASP #316: Lanbon L8 Screen Corruption / Unstable Firmware (0.6.2)

DJBenson, 2022-04-03:
### Perform all steps below and tick them with [x]
- [ X ] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [ X ] Update openHASP to the latest version
- [ X ] Reproduce the issue and describe all steps

### Describe the bug
Flashed a brand new Lanbon L8 over serial with the latest firmware, there is extensive screen corruption and the firmware is unusable on my device.

### To Reproduce
Flash new device using latest firmware
Configure device for WiFi and MQTT
Upload sample jsonl file for 3-relay example

### Expected behavior
Device works, screen is responsive and clear

### Screenshots or video
Will upload video after submission.

fvanroie, 2022-04-03:
What is the PCB version noted on the board, just below the pinheader?
If it is v1.17 it sounds a lot like [this frequency issue](https://github.com/HASwitchPlate/openHASP/discussions/65#discussioncomment-2100426).

DJBenson, 2022-04-03:
It is indeed v1.17

fvanroie, 2022-04-03:
I set the SPI frequency to 60MHz in the development branch.
Feel free to test it. The build is available at the [bottom of this page](https://github.com/HASwitchPlate/openHASP/actions/runs/2086040107).

DJBenson, 2022-04-03:
Thank you. I ran a custom build and amended the speed and it does indeed work fine. Thank you for confirming this issue and also for the fix.

fvanroie, 2022-04-03:
Thanks, this will be the new default setting in v0.6.3, since v1.17 PCB are getting more common these days.

closed 2022-04-03 by fvanroie