HASwitchPlate/openHASP #318: Screen rotation using Lanbon L8 does not affect touch

unrealmodder, 2022-04-12:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
Using latest build from source using GitPod.
If I rotate the display 90 degrees the touch interface remains at 0 degrees rotated whilst the UI is rotated 90 degrees. 

### To Reproduce
Use latest build from source then attempt to configure the rotation of the display through the web UI. 

### Expected behavior
Screen interface and touch overlay should rotate together.

fvanroie, 2022-04-12:
This should fix it, can you try again please?

closed 2022-04-19 by fvanroie

unrealmodder, 2022-04-26:
> This should fix it, can you try again please?

Hello, sorry for the delay but can confirm its fixed. Thank you!