HASwitchPlate/openHASP #319: Crash on D1 Mini ESP8266 all the time

galmiklos, 2022-04-12:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

I just flashed d1-mini-esp8266_ili9341_v0.6.1.bin (from https://github.com/HASwitchPlate/openHASP/releases/download/v0.6.1/d1-mini-esp8266_ili9341_v0.6.1.bin) on a D1 Mini ESP8266, but I couldn't configure the WiFi access through the web interface. So, I connected to the serial console, and configured ssid and pass. Apparently WiFi now works, but as soon as I try connecting to the plate with a web browser, it crashes.

Please, see the crash dump attached.

[openhasp_crash_dump.txt](https://github.com/HASwitchPlate/openHASP/files/8477031/openhasp_crash_dump.txt)

### To Reproduce

Reproducing the issue is trivial, any time I connect to the plate with a web browser, it crashes, and creates the crash dump on the serial console. Sometimes I can get as far as the main screen, but any button I click on results in a crash.

### Expected behavior

openHASP does not crash

### Screenshots or video

[image]

fvanroie, 2022-04-13:
From the logs it looks like memory exhaustion on the ESP. Only 3.5Kb of memory remains free after boot. The slightest workload will crash the system. From experience it needs about free 9Kb to run, but the ESP8266 is simply underpowered to handle all the services. Either some of the services need to be stopped/disabled or tweak the firmware to make additional memory available.

ESP8266 support is removed in the next release. It is just not worth the effort to keep supporting this chip due to its lack in resources. It is much easier to upgrade to a more powerful ESP32 for a few bucks. Hence there will be no more updates/fixes to the ESP8266 version for ESP8266.

You can try to upgrade to `v0.6.2` or downgrade to `v0.x` and check the memory after bootup to see if it has enough free RAM.
On ESP8266 it will always be a balancing act to manage the resources and if runs the performance will be horrible...

If you can, please upgrade to ESP32.

galmiklos, 2022-04-14:
Understood, and thank you for the quick response.

I have actually already purchased an ESP32, and everything works nicely on it! :-)

closed 2022-04-14 by fvanroie