HASwitchPlate/openHASP #331: MQTT Address Truncated

DJBenson, 2022-05-03:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug
The MQTT address field in restricted to 16 characters which leads to truncation of longer MQTT addresses. Clearly this is only an issue when addressing by DNS name rather than IP address but my server is 17 characters long so the last digit gets truncated if using the DNS name forcing me to use the IP address.

### To Reproduce
Enter an MQTT address longer than 16 characters, save the config then re-enter the MQTT config - characters > 16 will have been truncated.

### Expected behavior
Address is not truncated

In the below example the MQTT address was 'mqtt.longdomainname.co.uk'
[image]

fvanroie, 2022-05-03:
Thank you for reporting this issue. This bug has been confirmed in openHASP version 0.6.2 :  MQTT Server Domain gets truncated when set via web interface #304
 
It is resolved and [the bugfix](https://github.com/HASwitchPlate/openHASP/blob/master/CHANGLELOG.md#bug-fixes) will be included for release in v0.6.3.

DJBenson, 2022-05-03:
> Thank you for reporting this issue. This bug has been confirmed in openHASP version 0.6.2 :  MQTT Server Domain gets truncated when set via web interface #304
>
>
>
> It is resolved and [the [...]

Apologies, I searched for similar issues but seemingly filtered by closed so missed this fix.

fvanroie, 2022-05-03:
No worries! I'd rather have a bug reported twice than not at all 😆 
The new version is taking longer than anticipated, so yeah this was bound to happen.

closed 2022-05-03 by fvanroie