HASwitchPlate/openHASP #335: Support for backlight on Makerfab ESPTFT35(CA)

mmccool, 2022-05-09:
**Is your feature request related to a problem? Please describe.**
Currently openHASP control backlight on Makerfabs ESPTFT35(CA) board.  It's true that the standard board as-shipped has it disabled, BUT it can be enabled by soldering two pads together (marked R31 on the board; technically a 0 ohm resistor should be used, but a blob of solder or a small wire would also do the trick.  This disables TX in the UART and therefore serial debugging, but no big deal).  See https://github.com/Makerfabs/Project_Touch-Screen-Camera/issues/8

**Describe the solution you'd like**
A flag or some build-time configuration option (disabled by default) to allow backlight control using PWM on GPIO1.

**Describe alternatives you've considered**
Well, I could go in and try to hack the code myself, and then try to upstream it, but I'm lazy.

**Additional context**
This page should also be updated to note that backlight control can technically be added: https://openhasp.haswitchplate.com/0.6.2/devices/makerfabs-tft-touch/

fvanroie, 2022-05-10:
I discovered that resistor in the schematics but didn't find it very useful at the time. While technically it can be done, I haven't tested this as I'm not *that* precise with a soldering iron.

It's not just the debug output as flashing of the ESP32 over UART could be affected too. I'm not sure if that will still work after the resistor is soldered on. Granted that you can upload firmware via OTA but it makes recovery a lot harder. Have you tested flashing the ESP32 with the blob on?

Before delving into the code, I would suggest:
- Please try [openHASP 0.6.3-rc1](https://github.com/HASwitchPlate/openHASP/discussions/334). It has the option to disable the Serial Port from the Debug Settings
- The ESP32 will still output some bootloader junk to the pin on boot, but that shouldn't be an issue...
- There is no Web UI option to set `GPIO1` for backlight atm, but you can use the `config/gui` commands:
- Publish `{"bckl":1}` to topic `hasp/<plate>/config/gui` or `api/config/gui` [REST endoint](https://github.com/HASwitchPlate/openHASP/issues/246#issuecomment-1024829360)
- Reboot

I haven't tried *the blob*, but that could work. The Web UI won't reflect the selected backlight pin, but that's window dressing which we can address if it works.

fvanroie, 2022-05-26:
Did you have a chance to test UART with the backlight blob on?

closed 2022-06-01 by fvanroie