HASwitchPlate/openHASP #342: Crashloop if using img with src set.

DolceTriade, 2022-06-06:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

openHasp crash loops upon the addition of 

### To Reproduce

### Expected behavior
Not crash

### Screenshots or video

```
        open____ _____ _____ _____
          |  |  |  _  |   __|  _  |
          |     |     |__   |   __|
          |__|__|__|__|_____|__|
        Home Automation Switch Plate
        Open Hardware edition v0.6.4-dev

[05:44:38.702][110580/168024 34][    0/    0  0] DBUG: Started @ 115200 bps
[05:44:38.713][110580/168024 34][    0/    0  0] DBUG: Environment: ttgo-lilygo-pi_st7796
[05:44:38.726][110580/168024 34][    0/    0  0] UART: Started
[05:44:38.737][110580/168024 34][    0/    0  0] UART: Client login from serial
[05:44:38.749][110580/168024 34][    0/    0  0] CONF: SPI flash FS mounted
[05:44:38.787][110580/168024 34][    0/    0  0] CONF: Loading /config.json
[05:44:38.801][110580/167388 33][    0/    0  0] CONF: Loaded /config.json
[05:44:50.233][110580/168284 34][    0/    0  0] LVGL: Starting...
[05:44:50.252][110580/168092 34][62320/62544  1] LVGL: LV_FS_PC is initialized with.
[05:44:50.275][86004/142936 39][61620/61668  1] GUI : Initialize Cursor
[05:44:50.288][86004/142936 39][61528/61552  1] DRVR: Touch SDA     : 21
[05:44:50.301][86004/142936 39][61528/61552  1] DRVR: Touch SCL     : 22
[05:44:50.313][86004/142936 39][61528/61552  1] DRVR: Touch freq.   : 400000
[05:44:50.325][86004/142936 39][61528/61552  1] DRVR: Touch address : 38
[05:44:50.870][86004/142432 39][61528/61552  1] DRVR: FT6336U touch driver started chipid: 100
[05:44:50.887][86004/142432 39][61288/61304  1] LVGL: Started
[05:44:50.898][86004/142432 39][61288/61304  1] ATTR: src/hasp/hasp.cpp 544 0
[05:44:50.911][86004/142432 39][61288/61304  1] ATTR: src/hasp/hasp.cpp 545 0
[05:44:50.927][86004/142432 39][60424/60464  1] HASP: Custom theme loaded
[05:44:50.956][86004/142432 39][59288/59288  0] HASP: Loading /pages.jsonl
[05:44:51.408][86004/136196 36][46676/46684  1] ATTR: Unknown Font ID 28
[05:44:51.456][86004/136196 36][45448/45456  1] ATTR: Unknown Font ID 28
Prompt > Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40166fd3  PS      : 0x00060d30  A0      : 0x8015814f  A1      : 0x3ffb2170
A2      : 0x3ffd51b8  A3      : 0x3ffb21a0  A4      : 0x00000006  A5      : 0x3ffb2532
A6      : 0x0000002e  A7      : 0x3ff96355  A8      : 0x00000000  A9      : 0x3ffb2170
A10     : 0x00000002  A11     : 0x3ffb217c  A12     : 0xffffffff  A13     : 0x00000000
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400903a1  LEND    : 0x400903b1  LCOUNT  : 0xfffffffd

Backtrace:0x40166fd0:0x3ffb21700x4015814c:0x3ffb21a0 0x40168011:0x3ffb21d0 0x4016806f:0x3ffb21f0 0x40157975:0x3ffb2240 0x40129dd1:0x3ffb2260 0x40129719:0x3ffb22b0 0x4012dd1d:0x3ffb22e0 0x4012efc3:0x3ffb2310 0x4012f15b:0x3ffb2370 0x400e1309:0x3ffb2390 0x400e6761:0x3ffb2600 0x400e6bc
5:0x3ffb2660 0x400e36a0:0x3ffb2710 0x400e74e2:0x3ffb2770 0x400d9765:0x3ffb27b0 0x400dac67:0x3ffb27d0 0x400eef8d:0x3ffb2800 0x4013ae76:0x3ffb2820

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12596
load:0x40080400,len:2916
entry 0x400805c
```

DolceTriade, 2022-06-06:
You can work around this by not setting the "src" by default and wait until you connect to WiFi and let MQTT set it for you, but if that's the case, we should probably document this.

Checking WiFi connection prior to starting the HTTP get request does fix the problem, but it leads to other errors and it not network agnostic, so maybe there is a better way to fix this.

nagyrobi, 2022-06-06:
Also with batch processing, you could add the remote image later, within the L:/online.cmd script. At boot you could pre-load it with a dummy local image stored in flash.

DolceTriade, 2022-06-06:
Yes, there are many ways to work around this, but it kinda sucks if you forget to workaround it and you're stuck crashlooping. Then you have no way to recover without a erasing your flash and starting over (or pushing a hacked build doesn't load src/img)

nagyrobi, 2022-06-06:
Yeah, been there...

fvanroie, 2022-06-11:
Bootloops are very annoying, so I'll look into this.

fvanroie, 2022-07-02:
Okay, so the crash occurs because the network isn't initialized yet when the `pages.jsonl` file was parsed.
Starting the network before HASP *"fixes"* the crash. The image isn't properly loaded and will display `No Data` on screen, but at least it doesn't crash now.

nagyrobi, 2022-07-02:
Is that `No Data` a string which can be grabbed from the translations?

fvanroie, 2022-07-02:
nope because that is an LVGL message when the image isn't valid.

nagyrobi, 2022-07-02:
I had similar issue in the past when jsonl was referencing an image on the flash filesystem which was missing, don't know if that's still a problem...

fvanroie, 2022-08-18:
@DolceTriade Were you able to test the new commit and can you confirm this issue is resolved please?

DolceTriade, 2022-08-19:
I'll try to update my device and get back.

fvanroie, 2022-09-19:
I'll consider this issue closed. Feel free to update it if your tests indicates there is an issue.

closed 2022-09-19 by fvanroie