HASwitchPlate/openHASP #348: ESP32 - ili9341 parallel - adc touch not working

Al3XKOoL, 2022-06-29:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
I can't get adc touch working.
Screen is working on parallel very whell with TFT_eSPI, with default lovyangfx i get white screen but it's another bug.

Board: esp-wroom-32
DIsplay: 2.8 ILI9341 8-bit parallel with 4-wire analog touch
https://www.velleman.eu/products/view/?id=435582

### To Reproduce
Config from r32-unoshield_ili9341_adc:
```json
[env:d1-r32-unoshield_ili9341_adc]
extends = esp32_4mb_v2
board = esp32dev

build_flags =
    ${env.build_flags}
    ${esp32.build_flags}
    ${esp32.no_ps_ram}

;region -- TFT_eSPI build options ------------------------
    ${esp32.vspi}        ; Use VSPI hardware SPI bus
    -D USER_SETUP_LOADED=1
    ;-D LGFX_USE_V1=1
    -D ESP32_PARALLEL=1
    -D ILI9341_DRIVER=1
    -D TFT_WIDTH=240
    -D TFT_HEIGHT=320
    -D TFT_BACKLIGHT_ON=0 ; At what level is the backlight on
    -D TFT_ROTATION=0 ; Use default, see TFT_ROTATION values
    -D TFT_BCKL=-1  ;None, configurable via web UI (e.g. 2 for D4)
    -D TFT_CS=12  ;  XM
    -D TFT_DC=14  ;  YP
    -D TFT_RST=13
    -D TFT_WR=27
    -D TFT_RD=26
    -D TFT_D0=22  ;  YM
    -D TFT_D1=21  ;  XP
    -D TFT_D2=25
    -D TFT_D3=15
    -D TFT_D4=2
    -D TFT_D5=4
    -D TFT_D6=16
    -D TFT_D7=17
    -D SD_CS=6
    -D SPI_FREQUENCY=40000000
    -D SPI_READ_FREQUENCY=20000000
    -D TOUCH_DRIVER=0x0ADC ; No touch driver found? No problem
    -D TOUCH_anDC=12 ; Analog pin to be connected to TFT_DC
    -D TOUCH_anWR=14 ; Analog pin to be connected to TFT_WR

;endregion

; -- Debugging options -----------------------------
   ;-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG

;region -- Library options -------------------------------
lib_deps =
    ${env.lib_deps}
    ${esp32.lib_deps}
    ${arduino_esp32_v2.lib_deps}
    ${tft_espi.lib_deps}
    ;${lovyangfx.lib_deps}

lib_ignore =
    ${env.lib_ignore}
    ${esp32.lib_ignore}
    ${arduino_esp32_v2.lib_ignore}
;endregion
```

Changed in sources: touch_driver_analog.h
```json
const int XP = TFT_D1, XM = TFT_CS, YP = TFT_DC, YM = TFT_D0; // ID=0x9341
```

### Expected behavior
Normal touch functions

Al3XKOoL, 2022-06-30:
Hi.

After wiring from GPIO12 to GPIO34 and GPIO14 to GPIO35 touch start working, it's strange because my first test i already make this wiring mod following the gpio setup from "r32-unoshield_ili9341_adc" without success.

Now the only change is

`    -D TOUCH_anDC=34 ; Analog pin to be connected to TFT_CS`
`    -D TOUCH_anWR=35 ; Analog pin to be connected to TFT_DC`

The problem now is (All tests are following the cursor):
From UP/RIGHT corner to DOWN/LEFT corner, calibration is perfect.
From UP/LEFT corner to DOWN/RIGHT corner, touch is inverted.

RIGHT = GO UP
LEFT    = GO DOWN
UP       = GO RIGHT
DOWN= GO LEFT

I'll try fix it by remap X/Y and report here.

Al3XKOoL, 2022-06-30:
All working now!

After remap XY on "touch_driver_analog.h"

> // REMAPED //
            data->point.x = map(tp.y, TS_BOT, TS_TOP, max_y, 0);
            data->point.y = map(tp.x, TS_LEFT, TS_RT, 0, max_x);
            //data->point.x = map(tp.x, TS_LEFT, TS_RT, 0, max_x);
            //data->point.y = map(tp.y, TS_BOT, TS_TOP, max_y, 0);

And remap max to min/min to max:

> //const int TS_LEFT = 560, TS_RT = 3670, TS_TOP = 3850, TS_BOT = 580;
// REMAPED //
const int TS_LEFT = 3670, TS_RT = 560, TS_TOP = 580, TS_BOT = 3850;

Now need some calibration but it works, not perfect because cursos some times jump but i think it's enought.

You can close the issue.

fvanroie, 2022-07-01:
I don't own this specific diplay brand/model, so I can't troubleshoot myself.
Thanks for solving and posting the fix, this can help future users of this device.

closed 2022-07-01 by fvanroie