HASwitchPlate/openHASP #366: File name cannot exceed 30 characters

markusressel, 2022-08-08:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
Editing the file contents of file with a name longer than 30 characters leads to a `400 BAD REQUEST` response from the webserver.

### To Reproduce
* Go to the file editor
* create a file named `/1234567890123456789012345.jsonl` (31 chars in total, ignoring the leading `/`)
* type in some content into the editor
* hit ctrl+s
* see error

### Expected behavior
File content should be saved.
The file name length seems arbitrary, but might have a reason?

Alternatively a more descriptive error message should be shown.

### Context

In the tool I wrote to manage my configurations, I want to organize files in subfolders.
When I upload these files to the plate, I just concatenate the path with underscores in between (`_`).
This can lead to longer filenames pretty quick, f.ex.: `/common_page_about_0_header.jsonl` (32 characters).
It would be possible for me to automatically trim down the file names, but that would also make debugging harder, so I wanted to make sure if this is intended behavior or not.

PS: Awesome project!! :heart:

fvanroie, 2022-08-09:
This is a remnant of the SPIFFS filesystem which only supports filenames up to 31 characters.
LittleFS doesn't have this limitation, so the filename length check is now obsolete.

fvanroie, 2022-08-18:
@markusressel Were you able to test the new commit and can you confirm this issue is resolved please?

markusressel, 2022-08-19:
No, I trust in your skillz 😄 
I am not that familiar with building stuff using platformio, but if you need a real world test I can try get a compile working and make sure. Reading your comment I thought the fix was simple enough to make the change without a test, or simply reopen the issue if it didnt work out as expected.

fvanroie, 2022-08-19:
No worries, I'll consider this solved. Feel tree to reopen this issue if you still notice an issue...

FYI You don't have to compile the latest code yourself. The latest *nightly development builds* can be downloaded from the [Actions](https://github.com/HASwitchPlate/openHASP/actions) tab for testing.

closed 2022-08-19 by fvanroie

markusressel, 2022-08-19:
Thx for the hint, I just uploaded the latest master to one of my devices and it seems to work as expected :+1: :nerd_face: