HASwitchPlate/openHASP #381: Testing of Ethernet build with W5500

nagyrobi, 2022-09-14:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Using [Ben Suffolk's Ethernet base](https://github.com/HASwitchPlate/openHASP/discussions/65?sort=new#discussioncomment-2928054) with a Lanbon L8 rev 1.17 (originally shipped with an AU dimmer base), the following behavior is observed (v0.6.4-dev_7f4e079):
- with Ethernet it connects with the same MAC as with the WiFi? Surprising.
- `%ip%` variable is not populated when node boots on the messagebox. Would be nice to have it like it was in the WiFi times.
- pinging the plate shows an increasing latency following a saw-tooth pattern. Starts from around 7-10ms and slowly increases each reply until 1000ms, when it suddenly drops back to around 10ms. Period is about 3-4 minutes.
- it's really slow: A single screenshot takes about 35 seconds to display in the browser. May induce problems as afaik the web page re-requests the screenshot every 10 seconds... Screen unresponsive while displaying the shot in browser.
- FTP service doesn't work, fails with error `Response:	425 No data connection` (FileZilla client), tried both active and passive connections (same plate with WiFi binary has FTP working)
- `/info` page is empty in most of the cases (I was able to display it after several reboots)
- parts of the web interface load incompletely, for example MQTT settings fields either remain empty, or they need seconds to get filled in. Same with Firwmare Upgrade page's OTA URL sometimes autocompletes with the last used URL sometimes not (maybe related to slowness).
- OTA is damn slow. OTA from URL is impossible to complete. Bar slowly to the end on the screen but the plate doesn't reboot, instead the web interface jumps back to main menu. Rebooting from there doesn't result in booting from the new firmware image. OTA from uploaded file is OK.

### To Reproduce

Built [locally](https://www.openhasp.com/0.6.3/compiling/local/) by uncommenting `lanbon_l8_eth` in `platformio_override.ini`.

### Expected behavior

N/A
### Screenshots or video

N/A

fvanroie, 2022-09-14:
> with Ethernet it connects with the same MAC as with the WiFi? Surprising.

The W5500 doesn't have an assigned MAC-address, so in this implementation it was opted to re-use the WiFi MAC address.
It is possible to change to [`esp_read_mac(mac, ESP_MAC_ETH)`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv425esp_efuse_mac_get_defaultP7uint8_t) instead, if this is required.

> it's really slow

Most of the issues seem to revolve around the speed of the interface. I noticed by default the ETHSPI is run at 12MHz by default. The W5500 chip supports up to 80MHz and so does the ESP32. Try adding `ETHSPI_CLOCK_MHZ` to the config and set it incrementally to `20`, `40`, `60` or `80` (unit is MHz so don't go beyond `80`).

See if this improves screenshot performance, Web UI issues and/or Ping replies.

> %ip% variable is not populated when node boots on the messagebox. Would be nice to have it like it was in the WiFi times.

This is not implemented at the moment. Currently `%ip%` and `%ssid%` are only available when using WiFi.

nagyrobi, 2022-09-15:
I've set `ETHSPI_CLOCK_MHZ` to `80`, plate boots but does not connect at all. Brick...

fvanroie, 2022-09-15:
How about 20, 40, 60 MHz?

nagyrobi, 2022-09-15:
I'll try in the week-end.

nagyrobi, 2022-11-17:
Well, unfortunately it still does, no real success achieved. Still very slow, far from usable.

fvanroie, 2022-11-17:
Thanks. Since this is use contributed code, I'll leave it to @bensuffolk to give some pointers here.
I'm currently busy sith S3 and Arduino_GFX integration, so this will be low priority at this time.

bensuffolk, 2022-12-11:
Sorry,  been a bit busy with other things, but am hoping to spent some time in the Christmas holidays back on this project.

I have a feeling I had a play around with different value for the spi speed and had issues with it running faster than 12. But everything seems ok for me at that speed. The only thing not working for me is the file viewer but I expect that’s a not enabled type thing I need to look at.

I will look over it all though and comment / make changes as needed shortly

fvanroie, 2022-12-18:
> Well, unfortunately it still does, no real success achieved. Still very slow, far from usable.

From this information, it seems W5500 on ESP32 is limited to ~10Mbit/s. Maybe it's possible to benchmark the speed with a sketch so there is actual data to compare. From what I gather, other chips provide better throughput but can pose other limitations (more pins, bigger physical size, etc).

A *perfect* solution seems hard to find.

nagyrobi, 2022-12-18:
10Mbit/s is more than enough for IoT applications in 99% of the cases. W5500 has thr big advantage that it requires less pins, and it's bus can be shared with other peripherals - which is a big advantage on high I/O pin number applications, which seem to be quite often.

bensuffolk, 2022-12-18:
I have another project which uses W5500 and ESP32 -WROVER-E and I did some testing (so I can see debug output etc, which I can't get on the lanbon). 

I can get the ETHSPI_CLOCK_MHZ up to 32 without causing timeout errors, but after that it always errors.

I have not yet tried to up the speed on the lanbon module, but you can clearly not get anywhere near 80.

fantasmisiciliani, 2023-04-02:
I built the 3ab88a3 for my WT32-SC01 Plus with a W5500 Lite and I don't have any of the mentioned issues, except the unpopulated %ip% variable. Tested only at 32Mhz. I will do some further test with higher frequencies and report here.

I soldered a small LDO board to the W5500 Lite and connected it to the 5V pin of the I/O connector to get 3.3V.

SPI2_HOST default pins are exposed in the I/O onboard connector, so no need of the GPIO Matrix, but I had to modify SPI pin definitions according to the ESP32-S3 pinout. 

Because of some difference in the Arduino framework between the ESP32 and the ESP32-S3, I had to define CONFIG_ETH_SPI_ETHERNET_W5500 and add the following files from the ESP-IDF v4.4 to compile successfully: esp_eth_mac_w5500.c, esp_eth_phy_w5500.c, w5500.h

I also added a static IP configuration in src/sys/net/hasp_ethernet_esp32.cpp.
Setting no gateway (0.0.0.0), the console doesn't show the ethernet interface's info and the web server doesn't start, but MQTT works. Is it mandatory? 

My build flags become:
```
    ${wt32-sc01-plus.build_flags}
    -D HASP_USE_WIFI=0
    -D HASP_USE_ETHERNET=1
    -D HASP_USE_ETHSPI=1
    -D CONFIG_ETH_SPI_ETHERNET_W5500=1
    -D ETHSPI_MOSI_GPIO=11
    -D ETHSPI_MISO_GPIO=13
    -D ETHSPI_SCLK_GPIO=12
    -D ETHSPI_INT_GPIO=14
    -D ETHSPI_CS_GPIO=10
    -D ETHSPI_CLOCK_MHZ=32
    -D ETHSPI_IP=192,168,117,45
    -D ETHSPI_GW=192,168,117,1
```

@fvanroie @bensuffolk would you fix ETHSPI for the S3 and add a static IP configuration?
I modified void ethernetSetup() as the following, but I'm not a developer:
```
void ethernetSetup()
{
#if HASP_USE_WIFI == 0
    // Need to make sure we get the Ethernet Events
    WiFi.begin();
    WiFi.mode(WIFI_OFF);
#endif

    WiFi.onEvent(EthernetEvent);
#if HASP_USE_ETHSPI > 0

    if(HASP_ETHERNET.begin(ETHSPI_MOSI_GPIO, ETHSPI_MISO_GPIO, ETHSPI_SCLK_GPIO, ETHSPI_CS_GPIO, ETHSPI_INT_GPIO,
                                  ETHSPI_HOST)) LOG_TRACE(TAG_ETH, F("ETHSPI Started "));

// When a static IP is defined and the other parameters are missing we set them to default values
#ifdef ETHSPI_IP
    IPAddress ip(ETHSPI_IP);
#ifdef ETHSPI_GW
    IPAddress gw(ETHSPI_GW);
#else
    IPAddress gw(0, 0, 0, 0);
#endif
#ifdef ETHSPI_NETMASK
    IPAddress subnet(ETHSPI_NETMASK);
#else
    IPAddress subnet(255, 255, 255, 0);
#endif
#ifdef ETHSPI_DNS1
    IPAddress dns1(ETHSPI_DNS1);
#else
    IPAddress dns1(0, 0, 0, 0);
#endif
#ifdef ETHSPI_DNS2
    IPAddress dns2(ETHSPI_DNS2);
#else
    IPAddress dns2(0, 0, 0, 0);
#endif
//  it fails without a 3s delay
    delay(3000);
    if(HASP_ETHERNET.config(ip, gw, subnet, dns1, dns2)) 
        LOG_INFO(TAG_ETH, "Successfully assigned static IP to SPI Ethernet.");
    else
        LOG_INFO(TAG_ETH, "Failed to set static IP for SPI Ethernet.");
#endif // ETHSPI_IP

#else
    if(HASP_ETHERNET.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLKMODE))
                                  LOG_TRACE(TAG_ETH, F("ETH Started "));
#endif

}
```

fantasmisiciliani, 2023-04-25:
Further test on my WT32-SC01 Plus with W5500 Lite, openHASP 0.7.0-rc4 with mods as above and Arduino framework v2.7.0 (because v2.8.0 included in last Tasmota's platform causes errors in LovyanGFX lib): no problem up to 60Mhz, all works as expected.
Starting from 62Mhz the MQTT client doesn't connect and I can't reach/ping the device.
Download speeds with a 6 MB file:
- 2 Mbps from FTP, same speed at 32 or 60 Mhz
- 6.7 Mbps from HTTP at 32Mhz, 7.4Mbps at 60Mhz

closed 2024-02-17 by nagyrobi