HASwitchPlate/openHASP #385: No wifi reconnect on lanbon_l8_ota_v0.6.4-dev_14ba13e.bin after wifi access point reboot.

juan11perez, 2022-10-01:
### Describe the bug
Following (wifi) access point reset, the plate will not reconnect to wifi. 

### To Reproduce
This is happening with the development (lanbon_l8_ota_v0.6.4-dev_14ba13e) bin. 
Install dev firmware. Disconnect wifi to plate, reconnect wifi.

### Expected behaviour
On (wifi) access point returning to operation, plate reconnects to wifi. This bug is not present on 6.3. i.e the plate will reconnect as it continuously reboots until re-connection. On 6.4dev the plate seems to remain idle.

fvanroie, 2022-10-01:
Please test commit 80bda2c from the [latest build](https://github.com/HASwitchPlate/openHASP/actions/runs/3164515352).

juan11perez, 2022-10-01:
@fvanroie 
I installed lanbon_l8_ota_v0.6.4-dev_80bda2c.bin. Disabled wifi, plate displayed 'offline'. Enabled wifi and the plate reconnected.
I think it's working well now.
Thank you very much

closed 2022-10-01 by juan11perez

fvanroie, 2022-10-02:
Thanks for the report and testing.

juan11perez, 2022-10-02:
@fvanroie thank you for this excellent component.