HASwitchPlate/openHASP #390: Support for a new board WT32-SC01 Plus

rudam, 2022-10-10:
Dear friends,
is it possible to add support for the WT32-SC01 Plus board, please?
(ref.: https://www.aliexpress.com/item/1005004267336768.html)

Many thanks.
Michal

rudam, 2022-10-10:
This https://github.com/Skorpi08/WT32-SC01-PLUS can probably help.

fvanroie, 2022-10-13:
I am aware of this device and the repo you referenced.

There are boards that we own and test with that become "supported" configurations and there are user provided configurations.
For the first, I need to be able to get experience with the hardware. The latter is open to everybody and submit a configuration.
Do you own the board and are you able to run/test _**any**_ third party sketches (not openHASP specifically)?

I have been working on initial ESP-S3 support, but yet for the WT32-SC01 Plus.
So at some point it will be added, just not sure when that will be.
Any help will be appreciated.

jesse-schein, 2022-10-13:
I just ordered the WT32-SC01 Plus myself.

I also ordered some other cheaper resistive touch board but strangely it uses both SPI buses one for screen and one for touch so only Loyvan GFX supports it and its painful so OpenHASP and Tasmota are both out for these.

I'm going try to get openhasp running on the SC01-Plus I think it should work fine and compile or does OpenHASP not work at all with ESP-S3? Isnt the s-3 just a slightly different ESP32?

fvanroie, 2022-10-13:
> or does OpenHASP not work at all with ESP-S3

openHASP can run on the S3 fine. I'm testing with the ESP32-S3-DecKitC. It does require some minor tweaks and modifications, but all-in-all it is very similar to the ESP32. Some adjustments are to be expected. So, the first steps for ESP32-S3 support are already underway.

Adding new devices are usually performed in steps:
- [ ] Inventory all pins, LCD and Touch hardware chips, schematics, etc.
- [ ] Test any customs TFT sketch on the devicen to test the hardware, drivers, etc...
- [ ] Test openHASP drivers and/or make adjustments

Check out a recent [discussion about the GS-T3E](https://github.com/HASwitchPlate/openHASP/discussions/383).

rudam, 2022-10-14:
Hi. Thanks for your prompt response. I do have the SC01 PLUS and I am able to cooperate on testing.

fvanroie, 2022-10-16:
First thing will be to test the Skorpi sketch from Github to see if that's running on the board.
It will verify the pins and PIO settings to run 3rd party code on the SC01 Plus.

rudam, 2022-10-17:
Hello, I just got the answer from Wireless-Tag regarding the documentation.

Following are links for the docs provided by them.
[https://doc.smartpanle.com/index.html
](https://doc.smartpanle.com/index.html)[https://doc.smartpanle.com/_static/espboard/ESP32S3/ZX3D50CE02S-USRC-4832/ZX3D50CE02S-USRC-4832.pdf](https://doc.smartpanle.com/_static/espboard/ESP32S3/ZX3D50CE02S-USRC-4832/ZX3D50CE02S-USRC-4832.pdf)

rudam, 2022-10-18:
Another related project... [WT32-SC01-PLUS-LVGL-IDF](https://github.com/sukesh-ak/WT32-SC01-PLUS-LVGL-IDF)

mattkerrison, 2022-10-19:
Tried building the plus firmware but getting these errors whilst compiling:

```
Compiling .pio\build\wt01_sc01-plus\src\font\ascii\robotocondensed_regular_48_ascii.cpp.o
src/drv/tft/tft_driver_lovyangfx.cpp: In function 'lgfx::v1::Bus_Parallel8* dev::init_parallel_8_bus(Preferences*, int8_t*, uint8_t)':
src/drv/tft/tft_driver_lovyangfx.cpp:88:17: error: 'struct lgfx::v1::Bus_Parallel8::config_t' has no member named 'i2s_port'; did you mean 'port'?
             cfg.i2s_port = I2S_NUM_1;
                 ^~~~~~~~
                 port
src/drv/tft/tft_driver_lovyangfx.cpp:88:28: error: 'I2S_NUM_1' was not declared in this scope
             cfg.i2s_port = I2S_NUM_1;
                            ^~~~~~~~~
src/drv/tft/tft_driver_lovyangfx.cpp:88:28: note: suggested alternative: 'GPIO_NUM_1'
             cfg.i2s_port = I2S_NUM_1;
                            ^~~~~~~~~
                            GPIO_NUM_1
src/drv/tft/tft_driver_lovyangfx.cpp:92:17: error: 'struct lgfx::v1::Bus_Parallel8::config_t' has no member named 'i2s_port'; did you mean 'port'?
             cfg.i2s_port = I2S_NUM_0;
                 ^~~~~~~~
                 port
src/drv/tft/tft_driver_lovyangfx.cpp:92:28: error: 'I2S_NUM_0' was not declared in this scope
             cfg.i2s_port = I2S_NUM_0;
                            ^~~~~~~~~
src/drv/tft/tft_driver_lovyangfx.cpp:92:28: note: suggested alternative: 'GPIO_NUM_0'
             cfg.i2s_port = I2S_NUM_0;
                            ^~~~~~~~~
                            GPIO_NUM_0
*** [.pio\build\wt01_sc01-plus\src\drv\tft\tft_driver_lovyangfx.cpp.o] Error 1
```

fvanroie, 2022-10-19:
> Tried building the plus firmware but getting these errors whilst compiling:

It is still a work in progress.

jesse-schein, 2022-10-20:
I should be getting mine in a few days, I'm confident I could get it to work.

I was able to get this board to run just fine, seems someone added the bigger version of it already.

The annoying part on this board is the usage of 2 SPI buses, one for screen and one for touch. Which means I had to setup lovyan to make it work and copy things over from the manufactuer's demos. I still need help finalizing some if define statements.

If someone could help that'd be cool, @fvanroie can you take a look? I got stuck switching the SPI bus with the if defines, I dont know how to get platformio to detect my defines from platform.ini.

https://github.com/jesse-schein/openHASP

https://www.aliexpress.us/item/3256804315935867.html?spm=a2g0o.order_list.0.0.48725e5b5Dz20e&gatewayAdapt=glo2usa&_randl_shipto=US

fvanroie, 2022-10-21:
> https://www.aliexpress.us/item/3256804315935867.html?spm=a2g0o.order_list.0.0.48725e5b5Dz20e&gatewayAdapt=glo2usa&_randl_shipto=US

This issue is for the SC01 Plus is from Wireless-Tag with ESP32-S3.
Please use a separate [hardware thread](https://github.com/HASwitchPlate/openHASP/discussions/categories/hardware) for the Sunton boards.

mattkerrison, 2022-10-22:
Those changes compile and upload to the board fine!
I can connect to the wifi and mqtt but it's not being discovered by HA. I tried sending the discovery command but still nothing. 

From memory the board firmware and HA hacs openhasp has to be the same version for it to work? Is there a value I can change? I'm not seeing anything in the logs yet

fvanroie, 2022-10-22:
Check this thread for [MQTT discovery](https://github.com/HASwitchPlate/openHASP/discussions/388#discussioncomment-3843246). In 0.6.4 there is a small change in the topic reply.
Either adjust the manifest file in HA or manually send the discovery payload to the old topic.

mattkerrison, 2022-10-23:
> @mattkerrison Thanks for testing!
> 
> Check this thread for [MQTT discovery](https://github.com/HASwitchPlate/openHASP/discussions/388#discussioncomment-3843246). In 0.6.4 there is a small change in the topic reply. Either adjust the manifest file in HA or manually send the discovery payload to the old topic.

That got it working thanks!
The plus model is lightyears better than the original WT32-SC01. The viewing angles are actually acceptable and the case it comes in is well designed for mounting. This is the display I've been waiting for!

rudam, 2022-10-25:
[SC01 Plus enclosure.zip](https://github.com/HASwitchPlate/openHASP/files/9860752/SC01.Plus.enclosure.zip)
[image]
[image]

If there is anybody interested in the enclosure for SC01 Plus, here is my own, printed by JLCPCB (PA12-HP Nylon).

fvanroie, 2022-10-25:
I've added this board to the automated build steps, so from now on it is included in the *nightly builds*.

The serial console is going to the USB-C port, but that's quite a pain to use because it is unmounted/mounted when the board reboots. So if anybody is able to figure out how to redirect the debug log to the RX/TX pins on the debug header, please let me know...

rudam, 2022-10-25:
Thanks a lot. Will test it tomorrow.

jesse-schein, 2022-10-25:
> I've added this board to the automated build steps, so from now on it is included in the _nightly builds_.
>
> The serial console is going to the USB-C port, but that's quite a pain to use because it [...]

I just got the board and flashed it, I can get it to display the setup page but it doesnt create a hotspot and cant seem to join my network....

Tried via on screen keyboard and the serial.

ONF: Failed to load /config.json
[     109.370][176116/202780 13][55816/55828  1] CONF: Loading EEPROM
[     109.372][176116/202780 13][55816/55828  1] CONF: Loaded EEPROM
[     109.373][176116/202844 13][55816/55828  1] CONF: Loaded /config.json
[     109.378][176116/202844 13][55816/55828  1] CONF: Failed to save /config.json
[     109.381][176116/205132 14][55816/55828  1] MQTT: Disabled
[     109.387][176116/205820 14][55756/55788  1] HTTP: Stopped
[     109.388][176116/205820 14][55756/55788  1] WIFI: Stopped
[     109.389][176116/205820 14][55756/55788  1] MSGR: Rebooting the MCU now!

fvanroie, 2022-10-26:
Erase the whole flash first before flashing the firmware.

fvanroie, 2022-10-26:
openHASP running on the SC01 Plus:
[image]

jesse-schein, 2022-10-26:
> Erase the whole flash first before flashing the firmware.

Did you use platformio or esp tool directly? I couldn’t get it erase with platformio I guess….

jesse-schein, 2022-10-26:
> > Erase the whole flash first before flashing the firmware.
> 
> Did you use platformio or esp tool directly? I couldn’t get it erase with platformio I guess….

Also what debugger do you have connected to the board there? Is that better than usb serial weirdness? Does OpenHasp have OTA?

rudam, 2022-10-26:
It looks like everything works flawlessly. I am happy now ;-) Btw this board is so fast on booting and WiFI connection. Thanks.

Zebble, 2022-10-26:
Can anyone share how best to flash the nightly onto a bare board?  I've tried the standard erase_flash/write_flash 0x0 using the "full" bin and all I get is a blank screen, no logs...

Zebble, 2022-10-26:
> Can anyone share how best to flash the nightly onto a bare board? I've tried the standard erase_flash/write_flash 0x0 using the "full" bin and all I get is a blank screen, no logs...

Nevermind.  That *did* seem to work.  Not sure why, but it was up and running when I came back to my PC!

htvekov, 2022-11-09:
> I've added this board to the automated build steps, so from now on it is included in the _nightly builds_.
>
> The serial console is going to the USB-C port, but that's quite a pain to use because it [...]

Any update on the log issue ?

fvanroie, 2022-11-15:
I haven't had the time to look into it. It shouldn't be to hard to create a test sketch and play with the option in Arduino to know which settings make a difference.

bwente, 2022-11-26:
I can't seem to flash using the webtools.

`Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.`

I was able to use chrome to flash LOLIN D32. How are you getting it into flash mode?

fvanroie, 2022-11-26:
How did you connect the board? The USB-C is directly connected to the ESP32-S3 without any TTL chip. In order to flash it over USB-C you still need to put the factory boot-loader into flash mode by grounding `GPIO0`. `GPIO0` is only broken out on the 7 pin DEBUG JST 1.25mm header. That header also breaks out `IO0`, `RX/TX`, `GND`, `EN` so it can be used with an external TTL too, as usual.

So, you need to pull `GPIO0` to `GND` during power-on to put the factory boot-loader into programming mode. I'm using the breakout board from WT but a normal JST 1.25mm cable will also do the trick. This can be fiddly if you don't have the propper tools, but the good news is that this only needs to be done once. After flashing openHASP with its packaged boot-loader any subsequent flashes should be possible over USB-C.

Either use the SC01 Plus Programmer or use a [JST 1,25mm cable](https://www.aliexpress.com/item/4001235017139.html) to create your own:
[image]
[image]

or use IC Test Hook Clamp Grabbers:
[image]

In absense of that, you can solder jumper wires:
[image]

bwente, 2022-11-26:
Thank you. Your instructions in the documentation were clear.

> Before starting the flash process you have to put the ESP into flash mode:
> 
> 1. Connect GPIO0 to GND before booting the device
> 2. Power-on or reset the ESP while GPIO0 is connected to GND
> 3. Start the flash process

I just couldn't figure out where the pins are. I can't find a pinout or datasheet with the pin labels. On your photo which one is `GPIO0` and `GND`?

fvanroie, 2022-11-26:
GPIO0 and GND are the two pins next to the SPK connector:

bwente, 2022-11-26:
That got me a bit further.

[image]

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4207908a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x17b8
load:0x403b6000,len:0xe28
load:0x403ba000,len:0x30d4
entry 0x403b6274
I (24) boot: ESP-IDF v4.4-367-gc29343eb94-dirty 2nd stage bootloader
I (25) boot: compile time 17:51:26
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00300000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=cbc24h (834596) map
I (225) esp_image: segment 1: paddr=000dbc4c vaddr=3fc984c0 size=03604h ( 13828) load
I (228) esp_image: segment 2: paddr=000df258 vaddr=40378000 size=00dc0h (  3520) load
I (232) esp_image: segment 3: paddr=000e0020 vaddr=42000020 size=79f68h (499560) map
I (315) esp_image: segment 4: paddr=00159f90 vaddr=40378dc0 size=0f700h ( 63232) load
I (328) esp_image: segment 5: paddr=00169698 vaddr=50000000 size=00010h (    16) load
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (348) spiram: Found 16MBit SPI RAM device
I (348) spiram: SPI RAM mode: sram 80m
I (348) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (353) cpu_start: Pro cpu up.
I (357) cpu_start: Starting app cpu, entry point is 0x4037947c
I (0) cpu_start: App cpu up.
I (593) spiram: SPI SRAM memory test OK
I (601) cpu_start: Pro cpu start user code
I (601) cpu_start: cpu freq: 240000000
I (602) cpu_start: Application information:
I (604) cpu_start: Project name:     s3-rgb-lvgl
I (610) cpu_start: App version:      1
I (614) cpu_start: Compile time:     Jun 17 2022 17:51:24
I (620) cpu_start: ELF file SHA256:  5200c3937be408d4...
I (626) cpu_start: ESP-IDF:          v4.4-367-gc29343eb94-dirty
I (633) heap_init: Initializing. RAM available for dynamic allocation:
I (640) heap_init: At 3FC9C840 len 000437C0 (269 KiB): D/IRAM
I (646) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (653) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (659) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (665) spiram: Adding pool of 2048K of external SPI memory to heap allocator
I (674) spi_flash: detected chip: generic
I (678) spi_flash: flash io: qio
W (682) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (695) sleep: Configure to isolate all GPIO pins in sleep state
I (702) sleep: Enable automatic switching of GPIO sleep configuration
I (709) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (724) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (734) uart: queue free spaces: 5
I (738) ESP32S3_LCD: lcd_buffer_size: 16000, lcd_dma_size: 4000, lcd_dma_node_cnt: 4
I (747) ESP32S3_LCD: lcd init ok
I (1190) lcd st7796: MADCTL=28
I (1211) MAIN: Normal Mode
```

fvanroie, 2022-11-27:
WT32-SC01 Plus needs the [0.6.4-dev branch firmware](https://nightly.openhasp.com/). 0.6.3 is not supported on the SC01 Plus.

closed 2022-12-01 by fvanroie

enrixp, 2023-02-12:
After flashing I can't connect to my windows anymore comport error and windows can't detect how to reset the chip to make it show on my com port

olagt, 2023-02-21:
Hi 
I can't make openHASP run on SC01 Plus 3.5 (Panlee) examples from 8ms.xyz work ok.
web installer 6.3 says unsupported, and 7.0 dev can flash it , but the device stays black and stays in reboot -loop 
I tried to compile master branch but after flashing, it reboots al the time. I don't have HW debugger and I can't see what might be the reason. 
I tried use branch 6.4-dev - but there is no tag or branch itself. ( That was suggested in this thread) 
Any ideas ?
Ola
Any ideas ?

htvekov, 2023-02-21:
> Hi I can't make openHASP run on SC01 Plus 3.5 (Panlee) examples from 8ms.xyz work ok. web installer 6.3 says unsupported, and 7.0 dev can flash it , but the device stays black and stays in reboot [...]

You are doing the initial serial flash with an external serial FTDI device connected to the debug connector, right ?

vcelak, 2023-02-22:
Hello, I got 20 pcs of wt32-sc01 plus boards from wireless tag directly. Boards are working fine with original 8ms demos but with openhasp they fail in watchdog boot loop. Any suggestions please?

htvekov, 2023-02-22:
Hmm... Don't know if latest nightly builds in general have issues ?
Saw some reports on Discord about issues with other devices nightly builds.

Anyway, I'm currently on commit 16c900d from early february on my wt32-01 plus. No issues at all here.

New boards directly from WT you say ??
Is the pcb revision still V13 220511 ? Hopefully they haven't revised the pcb layout and gpio pins used

fvanroie, 2023-02-22:
Thanks for reporting this issue. Please try the [latest build](https://github.com/HASwitchPlate/openHASP/actions/runs/4246851787) and report back.

Also, could you do a `esptool.py.exe -p COM9 flash_id ` and report the `Detected flash size`. Thanks!

vcelak, 2023-02-23:
Hi, PCB rev is 221104 (datasheet here: http://www.wireless-tag.com/wp-content/uploads/2022/11/WT32-SC01-Plus-V1.3-EN.pdf).

I tried latest build as requested and it is working. Reported flash size is 128Mbit, PSRAM is 16Mbit. When I flashed 16MB version (firstly erased whole flash), first boot reported:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x128
load:0x403c9700,len:0xb9c
load:0x403cc700,len:0x27e0
entry 0x403c98bc
IDF/components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
IDF/components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}

Any other next boot was OK. If I tried 8MB version, then there was no IDF error at anytime.

Thank you for making things working :-)

olagt, 2023-02-24:
> Thanks for reporting this issue. Please try the [latest build](https://github.com/HASwitchPlate/openHASP/actions/runs/4246851787) and report back.
> 
> Also, could you do a `esptool.py.exe -p COM9 flash_id ` and report the `Detected flash size`. Thanks!
Hi @fvanroie thank you for your branch , however it doesn't work :(
My chip is: 
`esptool.py v4.5-dev
Serial port COM7
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:c0:86:e4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4018
Detected flash size: 16MB
Flash type: quad (4 data lines)
Hard resetting via RTS pin...`

[image]

Examples from 8ms.xyz work fine..
Ola

olagt, 2023-02-24:
Hey ,
Actually, after reconnecting to power, or suddenly it started to display "welcome" screen !!!

Thank you ! :) 

Ola

adorobis, 2023-03-07:
Waiting for my order to arrive I've noticed that the board has a speaker socket. Is there any chance to make it work with OpenHasp?

adorobis, 2023-03-08:
> If there is anybody interested in the enclosure for SC01 Plus, here is my own, printed by JLCPCB (PA12-HP Nylon).

Hi @rudam , many thanks for sharing your design, I'm trying to order it from JLCPCB from the same material. I've received the following question from them:

_"As shown below, the wall thickness of the indicated red areas are very thin, which has a high risk of deformation, crack and damage during the printing process._

_Could we kindly know if the risks are acceptable for you ? Shall we proceed as per your files ?_

_If not, could you please kindly increase the partial wall thickness to 1.5 mm for better print effect ？then we will activate the ' replace file' button for you."_

As I have zero 3D printing experience, did you face the same questions and you took the risk or you've updated the design to their recommendations?

adorobis, 2023-03-14:
Check this build:
https://github.com/HASwitchPlate/openHASP/actions/runs/4246851787
There is an 8MB version in there

adorobis, 2023-03-23:
Need to share that. SC01 Plus is addressing the main flow of SC01 design - the LCD viewing angles. Now it is all good, also the placement of USB socket makes it easier to mount in a frame.
Flashing is super easy using the USB cable and the online openhasp flashing tool.

So I have now two of those (and a bunch of old SC01s to sell).
One mounted in a wooden frame made to order (originally for the SC01 model):
[image]
And the second one in a 3d printed case (project shared earlier in this thread):
[image]

AlexSaplacan, 2024-09-08:
These boards are fantastic; we are using them on a few projects, and I'd like to know if they can be set up to use wired data rather than WiFi. I would greatly appreciate some examples.
Thanks

fvanroie, 2024-09-08:
I have not tried it myself, but some people have created a custom config for the [Lanbon L8 paired with a W5500](https://github.com/HASwitchPlate/openHASP/discussions/65?sort=new#discussioncomment-2882446) ethernet module.
https://s.click.aliexpress.com/e/_DmyZsif

Maybe it is possible to use this module on the WT32-SC01 Plus too..