HASwitchPlate/openHASP #392: Wifi failure on wt32-sc01 and 0.6.3

DannyWebb, 2022-10-14:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Master and Release 0.6.3, WiFi radio fails to initialize on a new wt32-sc01. Using the prebuilt image, I am unable to find the access point. Using UI or serial console, I cannot attach to an existing WiFi network.
Using source code, and either trying default config, or hard setting SSID/pass, device remains unable to be discovered or connect.

Revert to 0.6.2, and the device is discoverable. OTA update to 0.6.3 then works and WiFi remains enabled.

### To Reproduce

See above

### Expected behavior

WiFi to be enabled on a clean install

### Screenshots or video
Not Available

fvanroie, 2022-10-17:
@DannyWebb Thanks for your report. I have tweaked the WiFi code which should fix this issue.
Can you please test if [this 0.6.4-dev build](https://github.com/HASwitchPlate/openHASP/actions/runs/3265882219) resolves the issue? Be sure to completely wipe the flash first.

fvanroie, 2022-10-25:
I will consider this issue closed. Feel free to reopen if needed.

closed 2022-10-25 by fvanroie