HASwitchPlate/openHASP #397: Dropdown menu object with 'auto' open are not aligned and will eventually crash the plate

htvekov, 2022-11-02:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug

FW: openHASP 0.6.4dev - 72224eb or earlier (tested appx. some five weeks back in time)
HW: 'old' WT32-sc01 and new sc32-01 device tested

Using 'auto opened' dropdown menu's connected with a tabview parent id object, results in random placed, not aligned dropdown menu's.
The issue arises when one dropdown menu is closed by command and next dropdown menu is auto opened as next command to plate.

Shifting back & forth continuously bethween the two connected dropdown menu's on the parent tab. will eventually at some point crash the plate.

### To Reproduce

**Plate jsonl
plate name used in config/automation `wt32_01`**

```json
{"page":2,"comment":"---------- Page 2 ----------"}
{"page":2,"id":0,"prev":2,"next":2}
{"page":2,"id":95,"obj":"tabview","btn_pos":1,"y":106,"h":255,"hidden":0}
{"page":2,"id":96,"obj":"tab","parentid":95,"text":"Sources"}
{"page":2,"id":97,"obj":"tab","parentid":95,"text":"Playlists"}
{"page":2,"id":98,"obj":"dropdown","parentid":97,"x":60,"y":10,"w":200,"h":50,"hidden":0,"show_selected":1,"direction":0,"max_height":180,"bg_color":"#000000","border_color":"#C7BAA7","border_width":0,"text_color":"#FFAC00","text_font":16,"options":"source#01\nsource#02\nsource#03\n"}
{"page":2,"id":99,"obj":"dropdown","parentid":96,"x":10,"y":10,"w":200,"h":50,"hidden":0,"show_selected":1,"direction":0,"max_height":180,"bg_color":"#000000","border_color":"#C7BAA7","border_width":0,"text_color":"#FFAC00","text_font":16,"options":"playlist#01\nplaylist#02\nplaylist#03\n"}
```

**CC plate config in HA config.yaml**

```yaml
wt32_01:
    objects:    
      - obj: "p2b98"
        event:
          "up":
            - service: mqtt.publish
              data:
                topic: hasp/wt32_01/command
                payload_template: >-
                  {"page":2,"id":98,"close":1}
       
      - obj: "p2b99"
        event:
          "up":
            - service: mqtt.publish
              data:
                topic: hasp/wt32_01/command
                payload_template: >-
                  {"page":2,"id":99,"close":1}
```

**HA automation.yaml**

```yaml
- id: hasp_sonos_source_list_change
  alias: hasp_sonos_source_list_change
  mode: single
  trigger:
    - platform: mqtt
      topic: hasp/+/state/p2b95
  condition:
  - condition: template
    value_template: "{{ trigger.payload_json.event == 'changed' }}"
  - service: mqtt.publish
    data:
      topic: >-
        hasp/{{trigger.topic.split('/')[1]}}/command
      payload: >-
        {% if trigger.payload_json.val == 2 %}
          {"page":2,"id":98,"close":1}
        {% elif trigger.payload_json.val == 0 %}
          {"page":2,"id":98,"close":1}
        {% else %}
          {"page":2,"id":99,"close":1}
        {% endif %}

  ### TEMP DELAY TO TEST DROPDOWN MENUS !!! ###
  #- delay:
  #    seconds: 1
   
  - service: mqtt.publish
    data:
      topic: >-
        hasp/{{trigger.topic.split('/')[1]}}/command
      payload: >-
        {% if trigger.payload_json.val == 2 %}
          {"page":2,"id":99,"close":2}
        {% elif trigger.payload_json.val == 0 %}
          {"page":2,"id":99,"open":1}
        {% else %}
          {"page":2,"id":98,"open":1}
        {% endif %}
```

### Expected behavior

Same behaviour as on 0.6.3 release which always aligns dropdown menu's. Also no crashes experienced at all on 0.6.3 release, even if plate is heavily 'abused' by a massive amount of repeated shifts - even if plate has not finalized drawing the full object.

If the one second delay in between the close and next open command in the automation is activated, then dropdown menu will align properly. But plate will still at one point crash when shifting between the two menus.

### openHASP error logs

Typical plate crash will most often show up as LVGL- or variants of Core 1 panic error. 

```
[1000D[0KPrompt > [1000D[9C[15:12:32.064][69620/72004  3][54976/55544  2] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":99,"close":1}
[1000D[0KPrompt > [1000D[9C[15:12:32.107][69620/72208  3][54652/55496  2] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":98,"open":1}
[1000D[0KPrompt > [1000D[9CGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```
```
[1000D[0KPrompt > [1000D[9C[11:33:00.375][69620/72332  3][53188/54068  2] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":98,"open":1}
[1000D[0KPrompt > [1000D[9C[11:33:03.051][67572/70576  4][53188/53468  1] MQTT PUB: p2b95 => {"event":"changed","val":0,"text":"Sources"}
[1000D[0KPrompt > [1000D[9C[11:33:03.094][69620/72332  3][53188/53464  1] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":98,"close":0}
[1000D[0KPrompt > [1000D[9C[11:33:03.134][69620/72468  3][52684/53420  2] LVGL: lv_obj_get_state
[1000D[0KPrompt > [1000D[9C[11:33:03.144][69620/72436  3][52684/53420  2] LVGL: NULL pointer (0x00000000)
[1000D[0KPrompt > [1000D[9C[11:33:04.137][69620/72332  3][52684/53420  2] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":99,"open":1}
```
```
[1000D[0KPrompt > [1000D[9C[11:20:17.256][69620/70568  1][52876/53440  2] MQTT PUB: p2b95 => {"event":"changed","val":1,"text":"Playlists"}
[1000D[0KPrompt > [1000D[9C[11:20:17.312][71668/72332  0][52452/52964  1] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":99,"close":0}
[1000D[0KPrompt > [1000D[9C[11:20:17.352][71668/72428  1][52452/53404  2] LVGL: lv_obj_get_child
[1000D[0KPrompt > [1000D[9C[11:20:17.361][71668/72428  1][52452/53404  2] LVGL: NULL pointer (0x00000000)
[1000D[0KPrompt > [1000D[9C[11:20:17.479][71668/72332  0][52452/53404  2] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":98,"open":1}
```
```
[1000D[0KPrompt > [1000D[9C[11:26:11.734][65524/70344  6][53208/53420  1] MQTT PUB: p2b95 => {"event":"changed","val":0,"text":"Sources"}
[1000D[0KPrompt > [1000D[9C[11:26:11.773][65524/72100  9][52576/52884  1] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":98,"close":0}
[1000D[0KPrompt > [1000D[9C[11:26:11.984][65524/72100  9][53000/53324  1] MQTT RCV: hasp/wt32_01/command = {"page":2,"id":99,"open":1}
[1000D[0KPrompt > [1000D[9CGuru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
```

### Screenshots or video

Misaligned dropdown menu:
[image]

Correctly aligned dropdown menu:
[image]

htvekov, 2022-11-09:
Plate crashes seems to have been eliminated completely with latest commit [9885149](https://github.com/HASwitchPlate/openHASP/commit/988514970cfa6143199283ce97e3d28c5efb681c)

fvanroie, 2024-04-07:
Is this issue still valid?

closed 2024-05-13 by fvanroie