HASwitchPlate/openHASP #400: Touch / Tap to Wake not functioning correctly

mattkerrison, 2022-11-07:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

As [mentioned in discord](https://discord.com/channels/538814618106331137/692393568266092617/1037695439480295424) Touch to Wake does not function properly on the WT32-SC01 Plus (and likely the standard WT32-SC01). Currently it does 'wake' the device but the first touch registers on whatever button is pressed which should not occur.

### To Reproduce

Wait until device goes idle and tap screen, will register a tap on a button on the screen instead of just waking device.

### Expected behavior

Once idle first tap should 'wake' screen setting idle flag false / increasing backlight, but should **not** register a button press (as you cannot see what you're pressing with screen off)

### Screenshots or video

mattkerrison, 2022-11-07:
Another user on discord added their feedback on the issue:

> If you tap the screen really fast, first touch actually works as expected and only wakes up the device. But a 'normal' touch (a few hundred milliseconds) will also activate the underlying object. Could be related to specific devices only where touch panels reports false releases. In that case some debouncing will have to be implemented.

htvekov, 2022-11-07:
Tested on both wt32-sc-01 and the wt32-sc-01 plus model. Working perfectly now it seems 🎉
Can't identify any issues at my end. No object types are publishing any touch event types at all, before first release and next touch/hold/whatever

closed 2022-11-08 by fvanroie