HASwitchPlate/openHASP #405: Can't compile 0.6.3

LuisThe0ne, 2022-12-27:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

When i try to build 0.6.3 using Gitpod(https://gitpod.io#https://github.com/HASwitchPlate/openHASP/tree/0.6.3) I get the following Error:

```
gitpod /workspace/openHASP $ platformio run
Processing az-touch-mod-esp32_ili9341_4MB (board: esp32dev; framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
/home/gitpod/.pyenv/versions/3.8.12/bin/python3 -m pip install dulwich --global-option="--pure"
Collecting dulwich
  Using cached dulwich-0.20.50.tar.gz (430 kB)
Requirement already satisfied: urllib3>=1.25 in /home/gitpod/.pyenv/versions/3.8.12/lib/python3.8/site-packages (from dulwich) (1.26.6)
Skipping wheel build for dulwich, due to binaries being disabled for it.
Installing collected packages: dulwich
    Running setup.py install for dulwich: started
    Running setup.py install for dulwich: finished with status 'error'
/home/gitpod/.pyenv/versions/3.8.12/lib/python3.8/site-packages/pip/_internal/commands/install.py:229: UserWarning: Disabling all use of wheels due to the use of --build-option / --global-option / --install-option.
  cmdoptions.check_install_build_global(options)
    ERROR: Command errored out with exit status 1:
     command: /home/gitpod/.pyenv/versions/3.8.12/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-49eg3nyl/dulwich_b170436ee05e47b8890326aebe498cc1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-49eg3nyl/dulwich_b170436ee05e47b8890326aebe498cc1/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' --pure install --record /tmp/pip-record-misnkfgj/install-record.txt --single-version-externally-managed --compile --install-headers /home/gitpod/.pyenv/versions/3.8.12/include/python3.8/dulwich
         cwd: /tmp/pip-install-49eg3nyl/dulwich_b170436ee05e47b8890326aebe498cc1/
    Complete output (6 lines):
    usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: setup.py --help [cmd1 cmd2 ...]
       or: setup.py --help-commands
       or: setup.py cmd --help
    
    error: option --pure not recognized
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/gitpod/.pyenv/versions/3.8.12/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-49eg3nyl/dulwich_b170436ee05e47b8890326aebe498cc1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-49eg3nyl/dulwich_b170436ee05e47b8890326aebe498cc1/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' --pure install --record /tmp/pip-record-misnkfgj/install-record.txt --single-version-externally-managed --compile --install-headers /home/gitpod/.pyenv/versions/3.8.12/include/python3.8/dulwich Check the logs for full command output.
WARNING: You are using pip version 21.2.4; however, version 22.3.1 is available.
You should consider upgrading via the '/home/gitpod/.pyenv/versions/3.8.12/bin/python3 -m pip install --upgrade pip' command.
*** Error 1
ModuleNotFoundError: No module named 'dulwich':
  File "/home/gitpod/.pyenv/versions/3.8.12/lib/python3.8/site-packages/platformio/builder/main.py", line 186:
    env.SConscript(item, exports="env")
  File "/home/gitpod/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/gitpod/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/workspace/openHASP/tools/auto_firmware_version.py", line 12:
    from dulwich import porcelain
========================================================================================================================================================================= [FAILED] Took 2.99 seconds =========================================================================================================================================================================

Environment                     Status    Duration
------------------------------  --------  ------------
az-touch-mod-esp32_ili9341_4MB  FAILED    00:00:02.990
=================================================================================================================================================================== 1 failed, 0 succeeded in 00:00:02.990 ===================================================================================================================================================================
gitpod /workspace/openHASP $ 
```
### To Reproduce

Try to compile the Firmware for the following override file:

```
; USAGE: Copy this file and rename it to platformio_override.ini
;
; The platformio_override.ini file is not overwritten or monitored by git
; ONLY edit platformio_override.ini to make local changes to the parameters

[platformio]
extra_configs =
    ; Uncomment or edit the lines to show more User Setups in the PIO sidebar
    ; user_setups/darwin_sdl/*.ini
[... 95 lines cut ...]
build_flags =
    ${env.build_flags}
    ${esp32.build_flags}
;region -- TFT_eSPI build options -----------------------------------
    ${lcd.lolin24}
    ${esp32.vspi}        ; Use VSPI hardware SPI bus
    -D TFT_DC=5
    -D TFT_CS=26 
    -D TFT_RST=-1        ; RST
    -D TFT_BCKL=22       ; None, configurable via web UI (e.g. 21)
    -D TOUCH_CS=17       ; (can also be 22 or 16)
; -- Options ----------------------------------------
    -D HASP_USE_TELNET=1
;endregion

;endregion
```

### Expected behavior

The firmware compiling flawless like 0.6.2 did.

### Screenshots or video

[image]

fvanroie, 2022-12-27:
The error indicates an issue when installing a python package: `    error: option --pure not recognized`
Try installing the `dulwich` package first without that option, before running `platformio run`:

```sh
python3 -m pip install dulwich
platformio run
```

LuisThe0ne, 2022-12-27:
Now it spits out the same error a few times
```
In file included from lib/freetype/include/freetype/config/ftconfig.h:42,
                 from lib/freetype/include/freetype/internal/ftdebug.h:29,
                 from lib/freetype/src/base/ftadvanc.c:19,
                 from lib/freetype/src/base/ftbase.c:21:
lib/freetype/devel/ftstdlib.h:37:10: fatal error: lv_fs_if.h: No such file or directory
```

fvanroie, 2022-12-27:
I think it's best to use the latest dev version instead: https://gitpod.io/#https://github.com/HASwitchPlate/openHASP/tree/master

LuisThe0ne, 2022-12-28:
Yeah, i thought about that too but is it stable / ready to use?
Or do you know when the next release is published, maybe I will wait.

fvanroie, 2022-12-30:
Both are quite stable, but there will sure be bugs. The limited time I can spend seems better suited towards the next release instead of troubleshooting older builds and packages...

LuisThe0ne, 2022-12-31:
That sounds logical.
Nevertheless Thanks, the firmware works really good.

closed 2022-12-31 by LuisThe0ne

darkman7hr, 2023-10-09:
I just cant compile this Master version.

`Processing az-touch-mod-esp32_ili9341_4MB (board: esp32dev; framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2023.08.01/platform-espressif32.zip)
----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
NotGitRepository: No git repository was found at .:
  File "/root/.platformio/penv/lib64/python3.9/site-packages/platformio/builder/main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "/root/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/tmp/openHASP-master/openHASP-master/tools/auto_firmware_version.py", line 35:
    BUILD_FLAGS=[get_firmware_commit_hash(),get_flash_size()]
  File "/tmp/openHASP-master/openHASP-master/tools/auto_firmware_version.py", line 22:
    r = Repo('.')
  File "/root/.platformio/penv/lib64/python3.9/site-packages/dulwich/repo.py", line 1141:
    raise NotGitRepository(
========================================== [FAILED] Took 0.94 seconds ==========================================`

jclark2019, 2025-04-14:
Issue strill exists with latest build  even after manually installing dulwich.  Has anyone been able to build for any envs?

fvanroie, 2025-04-14:
Did you download the zip file? That will explain the issue...

Please use `git clone` command to pull the sources locally.