HASwitchPlate/openHASP #407: When setting the maximum number of pages to 12, pagination with actions won't work

alvarolobato, 2022-12-29:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

When configuring more than 12 max pages, pagination by using actions won't work and will move to other pages, depending on the max number of pages.

This behaviour was tested with master and it was originally discovered in 0.6.2

### To Reproduce
- Override the max number of pages by editing your compination init, for instance `wt32-sc01.ini`(also tried with darwin) and add the following flag: -D HASP_NUM_PAGES=30
- Compile and upload
- Send the example bellow
- Pagination with actions won't work. Using the page command will.
- If changing the MAX pages back to 12, it will work properly.

Snipped of code to check the behaviour
```json
{"page":0,"id":10,"obj":"btn","action":"prev","x":0,"y":435,"w":107,"h":45,"bg_color":"#1c1c1c","text":"\uE141","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":32} 
{"page":0,"id":11,"obj":"btn","action":"back","x":107,"y":435,"w":107,"h":45,"bg_color":"#1c1c1c","text":"\uE2DC","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":32}  
{"page":0,"id":12,"obj":"btn","action":"next","x":214,"y":435,"w":107,"h":45,"bg_color":"#1c1c1c","text":"\uE142","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":32}

{"page":1,"comment":"---------- Page 1 ----------"}
{"obj":"btn","id":4,"x":5,"y":5,"w":230,"h":58,"bg_color":"#000000","border_color":"#FFAC00","border_width":2,"radius":10,"radius01":10,"radius02":10,"text":"Lights On","value_ofs_x":-85,"value_font":32,"value_str":"\uE6E8","value_color":"#B6B6B6","text_color":"#B6B6B6","text_font":24}

{"page":2,"comment":"---------- Page 2 ----------"}
{"obj":"label","id":8,"x":5,"y":5,"w":230,"h":30,"bg_color":"#000000","border_color":"#C7BAA7","border_width":0,"radius":10,"text":"Kitchen Dimmer","text_color":"#B6B6B6","text_font":24}

{"page":3,"comment":"---------- Page 3 ----------"}
{"obj":"label","id":14,"x":42,"y":10,"w":236,"h":30,"bg_color":"#000000","border_color":"#C7BAA7","border_width":0,"text":"Gold","text_color":"#C7BAA7","text_font":24}

```

### Expected behavior
The pagination buttons from page 0, should be able to move to next page, previous page and home.

alvarolobato, 2022-12-29:
The reason is that the action field has only 4 bits and it will overflow after 15, ie 12 pages and three actions. This filed needs to be increased if more pages are needed, changeling the define is not enough.

closed 2022-12-29 by alvarolobato

fvanroie, 2022-12-29:
You can’t just exceed the 12 page limit, there is more code that needs to be adjusted too… this is by design.
Actions are 4-bit integers, including `next`, `prev`, `back` and _no action_: totaling 16 possibilities.
Changing the number of pages requires more adjustments in the code.

What's the use-case for more pages? And which limit would you propose?

bqzero, 2024-07-23:
FYI is someone is looking for a workaround check this https://github.com/HASwitchPlate/openHASP/discussions/770#discussioncomment-10127360