HASwitchPlate/openHASP #418: T3E device craches when serial log is disabled

htvekov, 2023-01-07:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Using openHASP 0.6.4-dev 7229ce2 and disabling serial log in webinterface will crash openHASP (infinite bootloop)
Reflasing with serial interface is needed.

### To Reproduce

T3E device flashed with precompiled fw from github (artifact: globalsecurity)
In the webinterface choose config, debug, set serial log to disabled, save changes and reboot the device.

### Expected behavior

Plate will function normally and serial log will be disabled

### Screenshots or video

### Log clip:

```json
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420f7e36
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x118
load:0x403c9700,len:0xba4
load:0x403cc700,len:0x287c
SHA-256 comparison failed:
Calculated: dcda850773bd0403ff32ba85264cbafe78307d743827038a88dc1de0e8f234d6
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98c0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x4206d71c  PS      : 0x00060833  A0      : 0x82067618  A1      : 0x3fce2ce0  
A2      : 0x3fc983c8  A3      : 0x00000000  A4      : 0x3fcaed20  A5      : 0x0000abab  
A6      : 0x00060823  A7      : 0x00060820  A8      : 0x4c4b4000  A9      : 0x00000000  
A10     : 0x60000000  A11     : 0x007002b6  A12     : 0xfffff000  A13     : 0x0000abab  
A14     : 0x00ffffff  A15     : 0x00060820  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace:0x4206d719:0x3fce2ce00x42067615:0x3fce2d10 0x420630de:0x3fce2d30 0x420607d2:0x3fce2d90 0x42019881:0x3fce2dd0 0x4201d359:0x3fce2e00 0x4201d39b:0x3fce2e20 0x42016a03:0x3fce2e40 0x4201491f:0x3fce2e60 0x42015ca6:0x3fce2ea0 0x42015dcd:0x3fce2f20 0x420198dd:0x3fce3010 0x420632c6:0x3fce3030 

ELF file SHA256: 0000000000000000
```

htvekov, 2023-02-04:
Identical issue observed with Sunton 7" fw 0.7.0-dev 16c900d

Note: 
On a WT32-SC01 Plus with exact same fw serial logging can be disabled without issues.

```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420fa9ca
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x118
load:0x403c9700,len:0xba4
load:0x403cc700,len:0x287c
SHA-256 comparison failed:
Calculated: dcda850773bd0403ff32ba85264cbafe78307d743827038a88dc1de0e8f234d6
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98c0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x4206fbe4  PS      : 0x00060833  A0      : 0x82069ae0  A1      : 0x3fce2d10  
A2      : 0x3fc983b0  A3      : 0x00000000  A4      : 0x3fcb2bc8  A5      : 0x0000abab  
A6      : 0x00060823  A7      : 0x00060820  A8      : 0x4c4b4000  A9      : 0x00000000  
A10     : 0x60000000  A11     : 0x007002b6  A12     : 0xfffff000  A13     : 0x0000abab  
A14     : 0x00ffffff  A15     : 0x00060820  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace:0x4206fbe1:0x3fce2d100x42069add:0x3fce2d40 0x420655a6:0x3fce2d60 0x42062d82:0x3fce2dc0 0x4201a675:0x3fce2e00 0x4201fde1:0x3fce2e30 0x4201fe23:0x3fce2e50 0x42017e0b:0x3fce2e70 0x42015d5f:0x3fce2e90 0x42016fae:0x3fce2ee0 0x420170d3:0x3fce2f60 0x4201a6d1:0x3fce3090 0x4206578e:0x3fce30b0 
```

fvanroie, 2023-02-08:
I've pushed an update for this issue, test at your own risk. 😄

htvekov, 2023-02-08:
Tested on both the T3E and Sunton device with openhasp 0.7.0-dev 1b9135e fw
Fix confirmed ok for both devices, thank you.

closed 2023-02-08 by htvekov