HASwitchPlate/openHASP #422: Changing pages does not work through the touch screen - WT32 SC01 plus

mishomvp, 2023-01-29:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
I have created several pages, but when I want to change them using the touch screen, nothing happens.
I can switch them through the web or the home assistant.
Touchscreen works fine and I can turn devices on and off.
openHASP - 0.7.0-dev d664693

When I tap left,right or the home button I see this:

[23:36:14.842][75764/84228 10][43900/44400  2] MSGR: TEXT = unknown type
[23:36:14.845][75764/82644  8][43900/44400  2] MSGR: Command 'down' not found
[23:36:14.988][75764/84224 10][44396/44404  1] MSGR: TEXT = unknown type
[23:36:14.991][75764/82276  7][44396/44404  1] MSGR: Command 'up' not found
[23:36:17.008][75764/84224 10][43900/44404  2] MSGR: TEXT = unknown type
[23:36:17.011][75764/82276  7][43900/44404  2] MSGR: Command 'down' not found
[23:36:17.104][75764/84224 10][44396/44404  1] MSGR: TEXT = unknown type
[23:36:17.107][75764/82628  8][44396/44404  1] MSGR: Command 'up' not found
[23:36:18.209][75764/84224 10][44396/44404  1] MSGR: TEXT = unknown type
[23:36:18.211][75764/82628  8][44396/44404  1] MSGR: Command 'down' not found
[23:36:18.320][75764/84224 10][44396/44404  1] MSGR: TEXT = unknown type
[23:36:18.323][75764/82628  8][44396/44404  1] MSGR: Command 'up' not found
[23:36:20.769][75764/84224 10][43900/44404  2] MSGR: TEXT = unknown type
[23:36:20.772][75764/82276  7][43900/44404  2] MSGR: Command 'down' not found
[23:36:20.905][75764/84224 10][44396/44404  1] MSGR: TEXT = unknown type
[23:36:20.908][75764/82628  8][44396/44404  1] MSGR: Command 'up' not found

### To Reproduce

Widgets Demo using for test or openHASP Mushroom-like cards.
### Expected behavior

### Screenshots or video

fvanroie, 2023-01-29:
The [syntax for `action`](https://www.openhasp.com/0.7.0/design/objects/#actions) has changed in `0.7.0-dev`. Please try this instead:

```json
{"page":0,"id":1,"obj":"btn","action":{"up":"page prev"},"x":0,"y":420,"w":105,"h":60,"bg_color":"#2C3E50","text":"\uE141","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":32}
{"page":0,"id":2,"obj":"btn","action":{"up":"page back"},"x":107,"y":420,"w":106,"h":60,"bg_color":"#2C3E50","text":"\uE2DC","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":32}
{"page":0,"id":3,"obj":"btn","action":{"up":"page next"},"x":215,"y":420,"w":105,"h":60,"bg_color":"#2C3E50","text":"\uE142","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":32}
```

I'll see if we can add the legacy syntax too for backwards compatibility.

mishomvp, 2023-01-29:
Yes! It works!! Is there somewhere the changes that are in this version so I can read them?

fvanroie, 2023-02-13:
Check the [changelog](https://github.com/HASwitchPlate/openHASP/blob/master/CHANGLELOG.md) or [documentation](https://www.openhasp.com/0.7.0/).

closed 2023-02-13 by fvanroie