HASwitchPlate/openHASP #425: New UI object: Polygon

schafran, 2023-02-12:
Would be lovely to have the option to draw polygons on the screen (ideally clickable) to draw basic shapes (like floor plans).
One option would be to extend the line object with a fill property where the line is converted to a polygon by connecting start and end points and fills the enclosed area.

fvanroie, 2023-03-02:
Currently there is no such widget available in LVGL.
There might be in the future, with some improvements planned in [LVGL 9](https://github.com/lvgl/lvgl/issues/3676).

For now it is best to use the image widget for more complex shapes.

fvanroie, 2023-12-06:
Closing. It doesn't look like LVGL9 will implement such a widget.

closed 2023-12-06 by fvanroie