HASwitchPlate/openHASP #431: Sunton 7 inches screen just flicker after flashing

primeroz, 2023-02-17:
## Questions

Hi, i know this is in beta so not expecting anything but i thought i would bring up that flashing the Sunton screen 7 inches from at least `sunton-8048s070c_full_16MB_v0.7.0-dev_972ed65.bin`  up to the lastest version availble in nightly `sunton-8048s070c_full_16MB_v0.7.0-dev_aa1f4e7.bin` result in a `screen flickering` and nothing else. 

Sometimes i can see the openHasp logo 

I can erase the flash ( by running the command at the right itme ) and flash other images, i went back up to `972ed65` but they all have the same issue. 

The `Stock` UI was working so i don't think is a problem with the actual device 

Is there a known working version i can flash so i can try openHasp out ? 

thanks

primeroz, 2023-02-17:
I am really new to esp32 stuff. I was powering it from a usb hub connected to my computer right after flashing it. 

I will try with a wall socket adapter.

fvanroie, 2023-02-17:
The 7" is quite power hungry, so the hub might fail to provide the minimal power. A power brick or powered USB hub should be fine. Flaky USB cables can also pose an issue, try swapping that too if the issue persists.

primeroz, 2023-02-17:
you were right, moved to the powered usb hub and it works . sorry for opening such a stupid issue :pray:

closed 2023-02-17 by primeroz

fvanroie, 2023-02-17:
No worries. 😃 
Have fun with the device!