HASwitchPlate/openHASP #432: Page swipe stopped working in last dev.

exotsk, 2023-02-19:
{"page":0,"id":6,"obj":"obj","swipe":1,"x":0,"y":0,"h":50,"w":320,"opacity":0,"comment":"swipe-area-at-top"}  from manual no longer working

fvanroie, 2023-02-19:
Try

```json
{"page":0,"id":6,"obj":"obj","swipe":{"down":"page back"},"x":0,"y":0,"h":50,"w":320,"opacity":0}
```

This needs an update in the documentation.

fvanroie, 2023-02-19:
I have added the `"swipe":1` option again for backward compatibility, so no change is needed in the jsonl with latest build.

exotsk, 2023-02-19:
it`s work. thnx!
LR in negative =) 
"swipe":{"down":"page back","left":"page next","right":"page prev"}

fvanroie, 2023-02-19:
> LR in negative =)

It is the direction of the swipe, so swiping to the **left** will go to the **next** page.

exotsk, 2023-02-19:
but i have {"page":0,"id":6,"obj":"obj","swipe":{"down":"page back","right":"page next","left":"page prev"},"x":0,"y":0,"h":50,"w":320,"opacity":0} now and it works as it was

fvanroie, 2023-02-19:
I don't get it... That seems counterintuitive to me.

So if you are on page 1:
- Swiping right-to-left should go to page 2
- Swiping left-to-right should go to page 12

That's how it was [implemented before](https://github.com/HASwitchPlate/openHASP/blame/c296a5e568bb947f711646f4428b64651e63be0f/src/hasp/hasp_event.cpp#L377-L388).

htvekov, 2023-02-20:
> I don't get it... That seems counterintuitive to me.
>
> So if you are on page 1:
>
> * Swiping right-to-left should go to page 2
> * Swiping left-to-right should go to page 12
>
> That's how it was [...]

I experience the exact behaviour as @fvanroie describes with latest dev build commit 425e134

fvanroie, 2023-02-20:
Thanks @htvekov for testing.
@exotsk Can you try with just `"swipe":1` in the latest build? It should work like before the change.

exotsk, 2023-02-21:
{"page":0,"id":6,"obj":"obj","swipe":1,"x":0,"y":0,"h":50,"w":320,"opacity":0} in 425e134823b1fccf61f9a0eada9e160092f0ecab
from left to right is back; to left is next

htvekov, 2023-02-21:
> {"page":0,"id":6,"obj":"obj","swipe":{"down":"page back","right":"page next","left":"page prev"},"x":0,"y":0,"h":50,"w":320,"opacity":0}

Hmm... I don't get it 🤔
On my Sunton 7" device, with latest commit `425e134`, I have the the same behaviour with `"swipe":1` as this sniplet below. The exact opposite behavior as you describe. Do you by accident have two swipe objects defined with the second cancelling the first one out ? 

`{"page":0,"id":50,"obj":"obj","swipe":{"down":"page back","right":"page prev","left":"page next"},"x":0,"y":0,"h":200,"w":800,"opacity":0}`

htvekov, 2023-02-21:
> unfortunately I am concerned about another issue. therefore, the card is still on the table. I was never able to send any images from the web to the board. in whatever ways (MQTT, HA service)

Custom Component `push_image` service definitely work. I use it a lot on my pages 🙂
To keep it as simple as possible, have you tried the example data from HA' dev tools ?
Just add the target bit and fill in your openhasp entity_id and object page/id

[image]

exotsk, 2023-02-21:
sorry for offtopic but, very big thnax for answer
i have on plate 
{"page":6,"id":15,"obj":"obj","x":3,"y":48,"w":200,"h":200,"radius":6,"clip_corner":1,"hidden":0}
{"page":6,"id":16,"obj":"img","x":0,"y":0,"w":200,"h":200,"parentid":15,"src":"","auto_size":1}

in services service: 
openhasp.push_image
data:
  image: https://people.sc.fsu.edu/~jburkardt/data/jpg/lena.jpg
  obj: p4b17
  width: 128
  height: 128
target:
  entity_id: openhasp.plate35

No reaction. HA component 0.7  and main

MQTT hasp/plate35/command/p4b15.hidden 1,0 work (change visible of rectangle)

hasp/plate35/command/p4b16.src https://people.sc.fsu.edu/~jburkardt/data/jpg/lena.jpg no reaction
if i use imange in local http like  http://192.168.1.110:8099/detect.jpg
i have ->
[image]

exotsk, 2023-02-21:
Syslog
object
facility: "local0"
facilityCode: 16
severity: "info"
severityCode: 6
tag: "RCV"
timestamp: "2022-12-31T17:00:00.000Z"
hostname: "MQTT"
address: "192.168.1.227"
family: "IPv4"
port: 55872
size: 158
msg: "[92m[49140/50080  1][92m[35760/36356  2]p4b16.src = https://www.google.ru/images/branding/googlelogo/1x/googlelogo_color_272x92dp.png"
21.02.2023, 22:19:10[node: debug 15](http://192.168.1.110:1880/#)
ATTR : msg.payload : Object
object
facility: "local0"
facilityCode: 16
severity: "warning"
severityCode: 4
tag: "ATTR"
timestamp: "2022-12-31T17:00:00.000Z"
hostname: "<132>plate35"
address: "192.168.1.227"
family: "IPv4"
port: 55872
size: 75
msg: "[92m[47092/47912  1][92m[35760/36356  2]HTTP result -1"

exotsk, 2023-02-21:
By Service in HA
`21.02.2023, 22:21:16[node: debug 15](http://192.168.1.110:1880/#)
RCV : msg.payload : Object
object
facility: "local0"
facilityCode: 16
severity: "info"
severityCode: 6
tag: "RCV"
timestamp: "2022-12-31T17:00:00.000Z"
hostname: "MQTT"
address: "192.168.1.227"
family: "IPv4"
port: 55872
size: 154
msg: "[92m[47092/50320  6][92m[35760/36356  2]p4b16.src = https://mydomen:8123/api/openhasp/serve/0f85c33cbb25cf477da81e1ce761d6ea"
21.02.2023, 22:21:16[node: debug 15](http://192.168.1.110:1880/#)
ATTR : msg.payload : Object
object
facility: "local0"
facilityCode: 16
severity: "warning"
severityCode: 4
tag: "ATTR"
timestamp: "2022-12-31T17:00:00.000Z"
hostname: "<132>plate35"
address: "192.168.1.227"
family: "IPv4"
port: 55872
size: 75
msg: "[91m[34804/47740 27][92m[35760/36356  2]HTTP result -1"`

exotsk, 2023-02-21:
HTTP Image

21.02.2023, 22:24:39[node: debug 15](http://192.168.1.110:1880/#)
RCV : msg.payload : Object
object
facility: "local0"
facilityCode: 16
severity: "info"
severityCode: 6
tag: "RCV"
timestamp: "2022-12-31T17:00:00.000Z"
hostname: "MQTT"
address: "192.168.1.227"
family: "IPv4"
port: 55872
size: 113
msg: "[91m[18420/33144 44][92m[35352/36300  3]p4b16.src = http://192.168.1.110:8099/detect.jpg"
21.02.2023, 22:24:39[node: debug 15](http://192.168.1.110:1880/#)
LVGL : msg.payload : Object
object
facility: "local0"
facilityCode: 16
severity: "warning"
severityCode: 4
tag: "LVGL"
timestamp: "2022-12-31T17:00:00.000Z"
hostname: "<132>plate35"
address: "192.168.1.227"
family: "IPv4"
port: 55872
size: 102
msg: "[91m[16372/33032 50][92m[35352/36092  3]Image draw cannot open the image resource"

exotsk, 2023-02-21:
if i try open  api/openhasp/serve/46d881da21ed56fa4a2db5587517ebaa link in browser i see this
[image]

htvekov, 2023-02-21:
If you push image paths `directly` via mqtt to the plate, you'll have to supply the file in a supported binary format or as png. openHASP fw doesn't support other file formats.

Images will have to be places in a whitelisted HA folder in order for openHASP to be allowed to retrieve the file.
Your path seems somewhat strange (but could be ok if your config is correctly setup)
Normally all files to be shared to/from HA, are placed in the www folder in HA (`config/www/`)
That folder is usually shared as `local` folder. Example from HA dev service below.

service: openhasp.push_image
data:
  image: http://192.xxx.x.xx:8123/local/Mikkel.jpg
  obj: p2b40
  width: 355
  height: 355
target:
  entity_id: openhasp.sunton_01

One other thing. One of your paths is a https and the other is a http ?
Hopefully you've not enabled SSL on the HA side ?
https _**can**_ work and the openHASP fw supports it, but life is just much easier without SSL active in HA. Trust me ! 😉
If you have SSL active in HA, then turn it off when doing these tests.
Custom component uses the `internal_url` setting from HA to create the file path in the mqtt message. 

 ```yaml
 internal_url: http://192.xxx.x.xx:8123
  external_url: https://myduckdnsaddress.duckdns.org
```

fvanroie, 2023-02-21:
Please create a different discussion topic, closing this issue now.

closed 2023-02-21 by fvanroie