HASwitchPlate/openHASP #436: Only first img object is loaded

Asbjoern, 2023-03-02:
### Perform all steps below and tick them with [x]
- [ X] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [ X] Update openHASP to the latest version
- [ X] Reproduce the issue and describe all steps

### Describe the bug
When adding multiple img's only the first one is loaded. It does not matter if the src is http:// og L:/
There is nothing wrong with the images. If I switch them around again only the first one is loaded.

Version | 0.7.0-dev 265f9c4
Build DateTime | Feb 25 2023 16:40:45 UTC
Environment | wt32-sc01-plus_16MB

### To Reproduce
Provide a small, independent code sample that can be used to reproduce the issue.
Format the code like this:
```
{"page": 1, "id": 1, "obj": "img", "src": "http://10.0.1.1/openhasp/hk-Attic.png", "auto_size": 1,"h":240}
{"page": 1, "id": 2, "obj": "img", "src": "http://10.0.1.1/openhasp/hk-Bedroom.png", "auto_size": 1,"w":240, "y":240}
```

### Expected behavior
Expected both images to load

### Screenshots or video
[image]
```
[12:57:05.924][73716/89860 17][59244/59404  1] HASP: Loaded /pages.jsonl
[12:57:06.446][77812/92136 15][58276/58436  1] LVGL: Image draw cannot open the image resource
[12:57:06.448][77812/90540 14][58276/58436  1] LVGL: Image draw error
[12:57:06.602][77812/92136 15][58168/58328  1] LVGL: Image draw cannot open the image resource
[12:57:06.604][77812/90428 13][58168/58328  1] LVGL: Image draw error
[12:57:06.757][77812/92136 15][58168/58328  1] LVGL: Image draw cannot open the image resource
[12:57:06.759][77812/90540 13][58168/58328  1] LVGL: Image draw error
[12:57:06.913][77812/92136 15][58168/58328  1] LVGL: Image draw cannot open the image resource
[12:57:06.915][77812/90540 14][58168/58328  1] LVGL: Image draw error
[12:57:08.065][73716/90192 18][58276/58436  1] LVGL: Image draw cannot open the image resource
[12:57:08.068][73716/88608 16][58276/58436  1] LVGL: Image draw error
[12:57:08.259][73716/90192 18][58168/58328  1] LVGL: Image draw cannot open the image resource
[12:57:08.262][73716/88464 16][58168/58328  1] LVGL: Image draw error
[12:57:08.458][73716/90192 18][58168/58328  1] LVGL: Image draw cannot open the image resource
[12:57:08.461][73716/88496 16][58168/58328  1] LVGL: Image draw error
[12:57:08.678][73716/90192 18][58168/58328  1] LVGL: Image draw cannot open the image resource
[12:57:08.681][73716/88496 16][58168/58328  1] LVGL: Image draw error
```

htvekov, 2023-03-02:
> 

Hi' @Asbjoern

Well, could be either one or perhaps even two things - out of two 😉
I've experienced exactly same issue and same log entries.

LVGL is **_not_** a happy camper, and you're straining the mcu by applying multiple png images simultaniously. Images that the mcu must process 'on the fly'

1. Convert the images to binary images instead and upload these. Use the online LVGL tool [here](https://lvgl.io/tools/imageconverter)

2. You might also experience that you actually run out of dedicated LVGL Memory handling/shifting between several medium/large sized bg images. I bumped my LVGL memory stack to 80kb (default 64kb) and haven't had any LVGL memory issues since then. Note that this will 'cost you' on the general device memory stack, so use with care !

[image]

Asbjoern, 2023-03-03:
Thanks for your reply.
Converting to binary did the trick and will work for me.

markusressel, 2023-03-05:
Out of curiosity, when sending images to a plate via the home assistant integration, is it automatically applying the binary transformation?

I am asking because I have implemented a kind of "camera view" which updates every couple of seconds. This works well on the last 0.6.x version, but does not work at all on 0.7.0-dev. I have not yet investigated this further and will probably open a new issue if it persists after 0.7.0 release (or earlier if desired).

htvekov, 2023-03-07:
Yep, Custom Components serves the images as binary images

Anything in the logs when you issue these push_image commands ?
Check both HA- and openHASP plate logs

mvturnho, 2023-03-22:
Hi @htvekov ,

I have this same issue on a sc01 plus with 16Mb. I migrated a configuration from my wt32-sc01 to the sc01plus.
On the wt32-sc01 all works perfectly and on the sc01plus I have the no data issue.
I use 0.7.0-rc2 3ab88a3 version on both displays.

I thought these devices should have the same problem when running the same OpenHasp version

htvekov, 2023-04-01:
> I thought these devices should have the same problem when running the same OpenHasp version

Devices have diffents specs. Most wt32-sc01's have 4MB flash where the plus model has 8 or 16MB.
Both have 2MB of PSram.
The plus model has an ESP32-S3 where the original version has an ESP32.

The difference you experience could have numerous causes. Difference in memory mapping, partition scheme etc.
The fact your on same fw does not neccesarily mean that fw is using same partition scheme. If you've OTA updated from 0.6.3 or 0.6.4 to 0.7.x the original partition scheme will remain. Only serial flashing with 0.7.x fw will use the new partition scheme.

I have not enough insight in this to give more than hints to the root cause.
We need @fvanroie insight on this 🙂

fvanroie, 2023-04-17:
> Hi @htvekov ,
>
> I have this same issue on a sc01 plus with 16Mb. I migrated a configuration from my wt32-sc01 to the sc01plus. On the wt32-sc01 all works perfectly and on the sc01plus I have the no [...]

Hard to say without the pages.jsonl file and images...

fvanroie, 2023-04-19:
```json
{"page": 1, "id": 1, "obj": "img", "src": "http://10.0.1.1/openhasp/hk-Attic.png", "auto_size": 1,"h":240}
{"page": 1, "id": 2, "obj": "img", "src": "http://10.0.1.1/openhasp/hk-Bedroom.png", "auto_size": 1,"w":240, "y":240}
```

Do the images show correctly when you upload the images to flash and load them from there?

closed 2023-07-23 by fvanroie