HASwitchPlate/openHASP #441: Take a Screenshot

markusressel, 2023-03-05:
**Is your feature request related to a problem? Please describe.**
For showing off openHASP and configurations a user has come up with, it would be really nice to be able to take a screenshot on the device itself, instead of having to take a photo.

**Describe the solution you'd like**
Provide a button on the webserver to take a screenshot and download it immediately.

**Describe alternatives you've considered**
Provide a command to save a screenshot on the internal storage, allowing the user to download it using the API or web file browser.

**Additional context**
N/A

htvekov, 2023-03-05:
You can already download/copy an image directly from the web interface.
Just right click on the image to copy or download it.
[image]

[image]

markusressel, 2023-03-05:
Whoops not sure how I was able to miss that one! 😅 
Thx for the info! ❤️

closed 2023-03-05 by markusressel

fvanroie, 2023-03-05:
The link `http://<ip>/screenshot?q=0` should give you a BMP file back that can be saved and opened in any image editor.