HASwitchPlate/openHASP #444: MQTT Issue with Sunton Display

wilwizard1979, 2023-03-10:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://haswitchplate.github.io/openHASP-docs/)
- [x ] Update openHASP to the latest version
- [x ] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce
The only simple code I have in the Display is below (just for testing)....

```json
{
    "page": 1,
    "id": 10,
    "w": 240,
    "obj": "label",
    "text": "%hostname%"
}

{"comment":" ----------- Buttons ------------"}

{"page":0,"id":1,"obj":"btn","action":"prev","x":0,"y":429,"w":106,"h":50,"bg_color":"#2C3E50","text":"\uE141","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":28}
{"page":0,"id":2,"obj":"btn","action":"back","x":107,"y":429,"w":106,"h":50,"bg_color":"#2C3E50","text":"\uE2DC","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":38}
{"page":0,"id":3,"obj":"btn","action":"next","x":214,"y":429,"w":106,"h":50,"bg_color":"#2C3E50","text":"\uE142","text_color":"#FFFFFF","radius":0,"border_side":0,"text_font":28}

{"page":2,"comment":" ---- My Awesome Color Picker Layout ----"}
{"page":2,"id":1,"obj":"obj","x":40,"y":100,"w":160,"h":160,"radius":100,"opacity":100,"border_opa":160,"border_width":4,"comment":"touch-catcher"}

{"page":3,"comment":" ---- My Awesome Color Picker Layout ----"}
{"page":3,"id":1,"obj":"label","x":5,"y":5,"h":50,"w":50,"text":"Hello","enabled":true,"hidden":false}
{"page":3,"id":2,"obj":"btn","x":5,"y":90,"h":90,"w":100,"text":"World","enabled":false,"hidden":false}

```

-->

### Expected behavior

### Screenshots or video

fvanroie, 2023-03-10:
Did you install the [Home Assistant Community Store (HACS)](https://hacs.xyz/)?
Please install the [openHASP Custom Component](https://www.openhasp.com/0.7.0/integrations/home-assistant/howto/) using HACS.

Make sure to enable `Show Beta versions` and select version `0.7.0` of the openHASP component in HACS.
[image]

wilwizard1979, 2023-03-10:
Thanks fvanroie, I think I lost something that I had to install this new Custom Component.
I did install OpenHASP custom component at the version shown above (0.7.0) and HACS at version 1.31.0.
Now I have a new problem

After rebooting, the OpenHASP Integration was auto-discovered. When clicking to configure, I received the following message:
`the integration could not be loaded, try to restart home assistant.`
 
After clicking ok and re-starting Home Assistant, the integration looks like this:
[image]

wilwizard1979, 2023-03-12:
Hi @fvanroie, anything I could check or that I might be doing wrong?