HASwitchPlate/openHASP #448: Sunton ESP32-8048S050C don't boot

xtimmy86x, 2023-03-17:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://www.openhasp.com/)
- [X] Update openHASP to the latest version
- [X ] Reproduce the issue and describe all steps

### Describe the bug
After correctly flash the nightly firmware the screen don't boot and flash blank repeatly

### To Reproduce
Flash nightly firmware

### Expected behavior
The device boot up but don't do correctly

### Screenshots or video

closed 2023-03-17 by xtimmy86x

fvanroie, 2023-03-17:
@xtimmy86x Is it working and did you find the issue?

xtimmy86x, 2023-03-17:
> 

Yes, it was a bad power supply, i did the test with the panel connected to pc and doesn't boot,
but when i connect to an external power supply it boot correctly