HASwitchPlate/openHASP #449: Button text not displayed correctly when "id" property overlaps with another button, even though "page" is different

markusressel, 2023-03-19:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
When specifying two "Buttons" on separate pages but with the same ID, it seems like in some scenarios the second button displays the text of the first button, even though they should not override or inherit properties since they are on separate pages.

### To Reproduce
This is the button I used when I detected the issue.

```json
{"page": 2, "id": 20, "obj": "btn", "x": 374, "y": 43, "h": 64, "w": 96, "text": "Button Text", "bg_grad_dir": "0", "bg_color": "#558B2F", "bg_color02": "#33691E", "border_color": "#FF0000", "border_width": 2, "radius": 20}
```

This is the other button on a separate page, which should be independent, but somehow influences the text on the button above:

```json
{"page": 1, "id": 20, "obj": "label", "x": 9, "y": 48, "w": 48, "h": 48, "text": "\ue335", "align": "center", "text_font": 32, "text_color": "white", "border_width": 0, "border_color": "#FF0000"}
```

### Expected behavior

The text specified for the given page and id combo is displayed.

### Screenshots or video

**Note:** The button in question is **marked with a red border**.

With ID 20 (Icon of another Button on a different page is displayed instead of the text - **this is the WRONG behavior**):
[image]
[image]

With ID 60 (The correct text is displayed**this is the CORRECT behavior**):
[image]
[image]

[image]

fvanroie, 2023-03-19:
Is the behavior the same with 0.7.0-rc1?

> The configuration is deployed using [openhasp-config-manager](https://github.com/markusressel/openhasp-config-manager).

What does this do exactly? Push the JSONL via MQTT?
Can you please provide the MQTT and serial logs when the issue occurs?

markusressel, 2023-03-19:
> Is the behavior the same with 0.7.0-rc1?
I have not installed the rc1 yet, but I have another device with an earlier 0.7.0-dev version (I think I downloaded this on 26.02.23) and the issue does **not** occur on this version, so it might be fixed already.

> What does this do exactly? Push the JSONL via MQTT?

No, the files are uploaded via the HTTP API to the device storage, and the device is restarted (also via REST call).

> Can you please provide the MQTT and serial logs when the issue occurs?

Not that easily. As the issue does not occur on the dev version mentioned above, it might not be necessary anymore?

fvanroie, 2023-03-20:
There has been a lot of refactoring in 0.7.0 which is due to be released.
If it's not occurring in 0.7.0, then I'm inclined to close the issue as such.

markusressel, 2023-03-20:
Yes I thing that makes sense. If it comes up again I can open a new issue.

closed 2023-03-20 by markusressel