HASwitchPlate/openHASP #457: More LVGL memory on esp32s3 with psram

xtimmy86x, 2023-03-30:
Hi there,

it would be possible to increase the amount of LVGL dedicated memory because sometimes 64KiB is small if i put some objects

Chorty, 2023-03-31:
//#define LV_MEM_SIZE (64 * 1024U)                    // 64KiB of lvgl memory (default 48)
//#define LV_VDB_SIZE (32 * 1024U)                    // 32KiB of lvgl draw buffer (default 32)
 isn't this what you are looking for? It is in the user override config

htvekov, 2023-03-31:
Default LVGL heap size is 64kb on higher spec devices.
I've tested a bit with 80kb LVGL heap. Not really sure if it's a good idea in the long run though, as you'll decrease the default program heap size equally. But it will cure issues with crashes and `LVGL: Out of memory, can't allocate a new buffer (increase your LV_MEM_SIZE/heap size)` errors in the openHASP log.

It's also possible to remove the specific LVGL heap completely and use the ESP32 head instead with `LV_MEM_CUSTOM=1`

fvanroie, 2023-04-01:
> sometimes 64KiB is small if i put some objects

The current default settings are empirical values, so it all depends on what you are trying to do. So far, it hasn't really been a issue. Can you please share the `pages.jsonl` and your device. I'm curious as to your configuration when hitting this limit.

As @htvekov mentions, you can set the LVGL memory in your custom built firmware or disable it and use heap memory instead.

closed 2023-05-09 by fvanroie