HASwitchPlate/openHASP #473: Need help with push_image

bwente, 2023-04-24:
I am getting the following error trying to troubleshoot the problem. I think I have everything correct.

**Failed to call service openhasp.push_image. Unknown error**

```
service: openhasp.push_image
data:
  obj: p1b26
  width: 200
  height: 200
  image: https://people.sc.fsu.edu/~jburkardt/data/jpg/lena.jpg
target:
  entity_id: openhasp.wt32

```

```
{"page":1,"id":25,"obj":"obj","x":3,"y":48,"w":200,"h":200,"radius":6,"clip_corner":1,"hidden":1}
{"page":1,"id":26,"obj":"img","x":0,"y":0,"w":200,"h":200,"parentid":25,"src":"","auto_size":1}
```

[image]

htvekov, 2023-04-24:
I'm not sure that openHASP 0.6.3 and corresponding Custom Component version actually supports push_image ?
Try and update to CC v0.7.0 and latest openHASP rc4

bwente, 2023-04-25:
I used the example from 0.6.3 documentation. I also have been able to load a png and it works.

I found this in the log. It is probably an issue with duckdns.

```
  File "/config/custom_components/openhasp/__init__.py", line 616, in async_push_image
    f"{get_url(self.hass, allow_external=False)}/api/openhasp/serve/{image_id}"
  File "/usr/src/homeassistant/homeassistant/helpers/network.py", line 200, in get_url
    raise NoURLAvailableError
homeassistant.helpers.network.NoURLAvailableError
```

The link does render when I visit in a browser.
`entity_picture: /api/media_player_proxy/media_player.juke?token=478314232a924ce11fd15d950cd5944741b8a11ab3590eca6ce94ce8ef4eabc9&cache=aa64bcd05e64a991`

bwente, 2023-04-26:
I fixed it for now. Thanks for your help.

I changed my "local" network to my duckdns url.

bwente, 2023-04-26:
[image]

closed 2023-04-26 by bwente

htvekov, 2023-04-29:
> I fixed it for now. Thanks for your help.
> 
> I changed my "local" network to my duckdns url.

Ohh, You're using DuckDNS and SSL certs in HA ?
Can't really recommend this, as config at best will work, but all your local, internal traffic will be passed through the internet (duckDNS)

I had serious issues getting this config to work and have abandoned it completely now.
https://github.com/HASwitchPlate/openHASP-custom-component/issues/81

bwente, 2023-04-30:
Yes, I think I will change it back. I have most of my stuff is local, I have Rhasspy running for voice. I wanted album off, just to show off.