HASwitchPlate/openHASP #474: Extend actions objects can perform

schneis, 2023-04-25:
I'm new to openHASP and so far am very impressed with the project - great work!

**Is your feature request related to a problem? Please describe.**
Feature Request

**A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]**
The "action" that an object can perform is currently just "p1" thru "p12", "prev", "next", & "back".
I would like to be able to define an object which can perform other actions local to the device.  

**Describe the solution you'd like**
Ideally, the object [action](https://www.openhasp.com/0.6.3/design/objects/#actions) would be able to perform any of the currently defined [commands](https://www.openhasp.com/0.6.3/commands/) which can be issued via MQTT/HTTP/Telnet.  Many of these actions do require additional parameters which would either need to be embedded with the object definition or perhaps just implement the [run](https://www.openhasp.com/0.6.3/commands/#run) command so that the contents of the .cmd or .jsonl specified as the action can be run when an object is clicked?

**Describe alternatives you've considered**
The only other workaround I can see is to implement this externally using MQTT, but that defeats the point of having the device handle these actions itself.

**Additional context**
If anything with this isn't clear please let me know and I'll do my best to elaborate.

Thank you!

fvanroie, 2023-04-25:
In 0.7.0-rc4 the action and swipe properties have been [changed](https://www.openhasp.com/0.7.0/design/objects/#actions) to allow any command. Check out the updated [Page Navigation example](https://www.openhasp.com/0.7.0/examples/pagination/).

Please try flashing 0.7.0-rc4 from the [nightly build](https://nightly.openhasp.com/).

schneis, 2023-04-26:
Fabulous - that looks like it will do exactly what I was asking - thank you!

I'll see about getting my firmware upgraded and testing this out.

closed 2023-04-30 by fvanroie