HASwitchPlate/openHASP #489: Support for a new board sunton ESP32-2432S032N/R/C

ComfortHouseSystems, 2023-05-22:
## Questions

Dear friends,
is it possible to add support for the sunton ESP32-2432S032N/R/C board?

fvanroie, 2023-05-22:
Sure, you can copy and adjust the configuration files from e.g. the 2.8 model and compile locally.

The download link for the documentation and examples isn't working for me, so you need to find out the hardware pins or ask the seller for the documentation.

ComfortHouseSystems, 2023-05-22:
Hi, Friend!
Unfortunately, I still can't compile locally - even standard configurations. I am very little familiar with "platformio".
The full archive at the seller's link is not included in size, I am attaching an archive with a diagram and examples.
[esp32-2432s032.zip](https://github.com/HASwitchPlate/openHASP/files/11531600/esp32-2432s032.zip)

0WL007, 2023-05-24:
Hello! 
This is my config file for esp32-2432s032c and esp32-2432s032r(r - not tested!)

Put the file esp32-2432s032.ini in \user_setup\esp32 and add the lines esp32-2432s032r_4MB and esp32-2432s032c_4MB to the section [override] - "extra_default_envs =" in file platformio_override.ini
<details>
  <summary>esp32-2432s032.ini</summary>

```

;***************************************************;
;       Sunton ESP32-WROOM custom dev board with    ;
;               - ST7796 TFT SPI 4-WIRE             ;
;               - XPT2046 or GT911 touch controller ;
;       esp32-2432s032c and esp32-2432s032r         ;
;***************************************************;

[esp32-2432s032]
extends = arduino_esp32_v2
board = esp32dev

build_flags =
    ${arduino_esp32_v2.build_flags}
    ${esp32.no_ps_ram}
    ${esp32.hspi}        ; Use HSPI hardware SPI bus
    -D LGFX_USE_V1=1
    -D ST7789_DRIVER=1
    -D TFT_WIDTH=240
    -D TFT_HEIGHT=320
    -D TFT_ROTATION=0          ; see TFT_ROTATION values
    -D TFT_RGB_ORDER=0         ; Colour order Blue-Green-Red
    -D TFT_SDA_READ            ; Read from display, it only provides an SDA pin
    -D TFT_RST=35        
    -D TFT_SCLK=14       
    -D TFT_DC=2          
    -D TFT_CS=15         
    -D TFT_MOSI=13       
    -D TFT_MISO=12       
    -D TFT_BCKL=27
    -D SUPPORT_TRANSACTIONS
    -D SPI_FREQUENCY=60000000
    -D SPI_READ_FREQUENCY=6000000 

;region -- touch driver for GT911 ------------------------
[touch-GT911]
build_flags =
    -D TOUCH_DRIVER=0x0911 ; GT911
    -D HASP_USE_LGFX_TOUCH=1
    -D TOUCH_SDA=33
    -D TOUCH_SCL=32
    -D TOUCH_IRQ=-1   ; not connected (or 36)
    -D TOUCH_RST=25   ; 
    -D TOUCH_WIDTH=240
    -D TOUCH_HEIGHT=320
    -D I2C_TOUCH_FREQUENCY=400000
    -D I2C_TOUCH_PORT=1
    -D I2C_TOUCH_ADDRESS=0x5D ; or 0x14
;endregion

;region -- touch driver for XPT2046 ------------------------
[touch-XPT2046]
build_flags =
    -D TOUCH_DRIVER=0x2046 ; XPT2064
    -D SPI_TOUCH_FREQUENCY=2500000
    -D HASP_USE_LGFX_TOUCH=1
    -D TOUCH_SCLK=14
    -D TOUCH_CS=33
    -D TOUCH_MISO=12
    -D TOUCH_MOSI=13
    -D TOUCH_IRQ=36
;endregion

;region -- sunton-led.build_flags ------------------------
[sunton-led] ;GPIO inverted
build_flags =
    -D LED_RED=4
    -D LED_GREEN=16
    -D LED_BLUE=17
;endregion

;region -- sunton-sdcard.build_flags for future use -------
[sunton-sdcard]
build_flags =
    -D HASP_USE_SDCARD=1
    -D SD_MOSI=23
    -D SD_MISO=19
    -D SD_SCK=18
    -D SD_CS=5
;endregion

;region -- sunton-audio.build_flags for future use with SD card-------
[sunton-audio]
build_flags =
    -D HASP_USE_I2S=1
    -D I2S_DAC_CHANNEL_LEFT_EN=2 ; Enable I2S built-in DAC left  channel, maps to DAC channel 2 on GPIO26
;endregion

; for future - Photoresistor gt36516 on IO34
; GPIO on board connector - P3: IO21,IO22,IO35 and GND. CN1: IO21,IO22,GND and 3.3V

[env:esp32-2432s032r_4MB] ; NOT TESTED!!!
extends = esp32-2432s032, flash_4mb
board = esp32dev

build_flags =
    -D HASP_MODEL="SUNTON ESP32-2432S032R 4MB"
    ${esp32-2432s032.build_flags}
    ${touch-XPT2046.build_flags}
    ${sunton-led.build_flags}

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG

lib_deps =
    ${arduino_esp32_v2.lib_deps}
    ${lovyangfx.lib_deps}

[env:esp32-2432s032c_4MB]
extends = esp32-2432s032, flash_4mb
board = esp32dev

build_flags =
    -D HASP_MODEL="SUNTON ESP32-2432S032C 4MB"
    ${esp32-2432s032.build_flags}
    ${touch-GT911.build_flags}
    ${sunton-led.build_flags}

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG

lib_deps =
    ${arduino_esp32_v2.lib_deps}
    ${lovyangfx.lib_deps}

```

</details>

0WL007, 2023-05-24:
[esp32-2432s032.zip](https://github.com/HASwitchPlate/openHASP/files/11550303/esp32-2432s032.zip)

relyd, 2023-06-19:
@0WL007  Here the default sunton binary via https://nightly.openhasp.com/ works fine for the 2432S028R  (R version), except for the touch that is not working. 

What board variant do you use? C or R ?

Interesting is that if I start the "calibrate" command, the touch does respond.

0WL007, 2023-06-21:
> @0WL007 Here the default sunton binary via https://nightly.openhasp.com/ works fine for the 2432S028R (R version), except for the touch that is not working.
> 
> What board variant do you use? C or R ?
Hi!
my board - sunton esp32-2432s**032**c

mninja79, 2023-07-01:
Hi, curious to know if any has powered this device using lithium battery. I plugged one in and the device will not turn on. When on usb power it works and there is 4.2 sent to the lithium input to charge the battery. Would this be a config issue or hardware.
I have seen with some devices that are powered from a battery need to be turned on.

fvanroie, 2023-07-03:
> Hello! This is my config file for esp32-2432s032c and esp32-2432s032r(r - not tested!)

Can you create a Pull Request for the new ini file please? Thanks.

0WL007, 2023-07-04:
@fvanroie 
> Can you create a Pull Request for the new ini file please? Thanks.
I see that the ini file for esp32-2432s032c has already been added by respected d00616. Should I do a pull request in this case? esp32-2432s032r - not tested.

fvanroie, 2023-07-04:
OK, thanks for that reminder :)
It that config working for you? So, can this issue be closed?

closed 2024-02-29 by fvanroie