HASwitchPlate/openHASP #509: Boot cycle when adding tabiew with btn_pos=3

akloeckner, 2023-07-29:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

Lanbon L8 enters boot cycle when adding `tabiew` with `btn_pos=3`.

### To Reproduce

* Flash L8
* Add the lines from the documentation, but switch `btn_pos`:
  ```
  {"page":1,"id":14,"obj":"tabview","btn_pos":3,"y":180}
  {"page":1,"id":51,"obj":"tab","parentid":14,"text":"Tab 1"}
  {"page":1,"id":52,"obj":"tab","parentid":14,"text":"Tab 2"}
  {"page":1,"id":53,"obj":"tab","parentid":14,"text":"Tab 3"}
  ```
* Observe the device rebooting repeatedly.

### Expected behavior

Device should not crash. Or at least should be recoverable wthout reflashing it.

### ~Screenshots or video~ Log

```
[...]
[       0.725][110580/168056 34][    0/    0  0] TFT : Started
[       1.138][110580/167956 34][    0/    0  0] LVGL: Starting...
[       1.157][110580/167764 34][62320/62544  1] LVGL: LV_FS_PC is initialized with.
[       1.181][86004/142608 39][61620/61668  1] GUI : Initialize Cursor
[       1.195][86004/142608 39][61528/61552  1] DRVR: Touch SDA     : 4
[       1.207][86004/142608 39][61528/61552  1] DRVR: Touch SCL     : 0
[       1.219][86004/142608 39][61528/61552  1] DRVR: Touch freq.   : 400000
[       1.231][86004/142608 39][61528/61552  1] DRVR: Touch address : 38
[       1.777][86004/142104 39][61528/61552  1] DRVR: FT6336U touch driver started chipid: 100
[       1.794][86004/142104 39][61288/61304  1] LVGL: Started
[       1.805][86004/142104 39][61288/61304  1] ATTR: src/hasp/hasp.cpp 544 0
[       1.818][86004/142104 39][61288/61304  1] ATTR: src/hasp/hasp.cpp 545 0
[       1.841][86004/142104 39][59360/59476  1] HASP: Custom theme loaded
[       1.868][86004/142104 39][58256/58296  1] HASP: Loading /pages.jsonl
Prompt > Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  1 register dump:
PC      : 0x401d1102  PS      : 0x00060b30  A0      : 0x80108e01  A1      : 0x3ffb2310
A2      : 0x3f43c075  A3      : 0x00000008  A4      : 0x00000005  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x0000000f  A8      : 0x01010dde  A9      : 0x3f43c07d
A10     : 0x00000001  A11     : 0x3ffb392a  A12     : 0x00000005  A13     : 0x3ffb2630
A14     : 0x00000032  A15     : 0x00322062  SAR     : 0x00000001  EXCCAUSE: 0x00000009
EXCVADDR: 0x3f43c081  LBEG    : 0x400903e5  LEND    : 0x400903f5  LCOUNT  : 0xfffffffe

Backtrace:0x401d10ff:0x3ffb23100x40108dfe:0x3ffb2330 0x401214f4:0x3ffb2350 0x400dca21:0x3ffb2370 0x400e0fb6:0x3ffb2390 0x400e6ca9:0x3ffb2600 0x400e710d:0x3ffb2660 0x400e3be8:0x3ffb2710 0x400e7a2a:0x3ffb2770 0x400d9cb9:0x3ffb27b0 0x400db1bb:0x3ffb27d0 0x400ef4e5:0x3ffb2800 0x4013b3ea:0x3ffb2820

ELF file SHA256: 0000000000000000

Rebooting...
eFuse Two Point: NOT supported
eFuse Vref: Supported
```

darkman7hr, 2023-12-16:
Was this working at some point? I need tabs to be vertical, I'm guessing this is way to go.
I think Tasmota firmware has recovery when device is rebooted x times it defaults to factory. Might not be bad idea to implement.

akloeckner, 2023-12-17:
I'm not sure. I think, I saw some vertical tabs on my device, but then I gave up, because of that issue. If you really need to have those, you might want to try and set them up dynamically at run time, i.e. via the jsonl command. That way, the device might reboot, but it will not enter the boot *loop*. If you use the HA custom component, that is most convenient anyways.

I resorted to having just the default configuration on the device itself. Just to avoid any risk. HA then sends the entire configuration to the device. That way, I can recover in case of any issues (without unmounting the device out of my wall 😄 ). But I have not tried vertical tabs again yet.

darkman7hr, 2023-12-17:
good idea :)
I wonder why it bricks tho. It shouldn't be very hard to fix it I think but I don't want to brick device once more and like you said, unscrew it from wall and everything,.

darkman7hr, 2024-01-15:
any updates on fix time? my device is just sitting there waiting..

AlexanderMohan, 2024-02-18:
just wanted to add that I am experiencing this too on an ESP32-Terminal, version 0.7.0-rc10

AlexanderMohan, 2024-02-18:
[image]
it's also not displaying my second tab, and instead doing something strange and inserting a random tab with the label 'tab' rather than '2'. Unsure how to get logs but will happily provide these.

regierbot, 2025-05-03:
Hello,

I just extended my OpenHASP configuration (which was running smoothly for ~2 years!) and accidentally ran into the same tabview/btn_pos=3 bug ending up in an endless loop of boot&crash. Of course, I did not backup my extensions (my bad :-/). Now, I tried flashing the ESP32 with the full firmware v0.6.3 (without erasing the MCU before). I was hoping to recover my configuration file. However, as the buggy config file is retained, I keep getting crashes when it is loaded.

Is there a way to recover my configuration file or to tell OpenHASP somehow to avoid loading that buggy config file (instead loading a default config file) until I can fix it? I know I can erase and start over, but then I would also loose my latest config file.

Thanks for any help!