HASwitchPlate/openHASP #513: MQTT Transport Errors on Lolin D32 Pro

jgoldwein, 2023-08-05:
### Perform all steps below and tick them with [x]
- [ X] Check the related part of the [Documentation](https://www.openhasp.com/)
- [X] Update openHASP to the latest version
- [ X] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

fvanroie, 2023-08-05:
What firmware version are you using?

Some suggestions:
- Erase the flash first, then re-upload the customized firmware
- If that doesn't work, erase the flash and upload the default firmware
- Make sure to update the `full` firmware binary (don't update from 0.6.x to 0.7.0 using OTA)

jgoldwein, 2023-08-05:
Thanks for quick response.   Great software by the way - thanks very much!!!!

Using latest release ([openHASP 0.7.0-rc7) and have flashed full version from command line as:
 "C:\Users\goldw\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.5.1/esptool.exe" --chip esp32 --port "COM7" --baud 115200  --before default_reset --after hard_reset write_flash -e -z --flash_mode dio --flash_freq 40m 0x0 "C:\Users\goldw\OneDrive\Desktop\tasmota\HAWP1P1-plate02-251-full.bin"

Note that the issue is resolved with the following changes:

--- a/user_setups/esp32/_esp32.ini
+++ b/user_setups/esp32/_esp32.ini
@@ -140,7 +140,7 @@ lib_deps =
 extends = esp32
 framework = arduino
 ; platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.01.01/platform-espressif32.zip
-platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.06.04/platform-espressif32.zip
+platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.05.01/platform-espressif32.zip

 lib_ignore =
     ${esp32.lib_ignore}

jgoldwein, 2023-08-05:
PS - If you do not have a D32 Lolin Pro I am happy to send one to you.

fvanroie, 2023-08-06:
Great to hear it works by downgrading the Tasmota Arduino Core, so must be an issue in the platform code.

Meanwhile [Tasmota Arduino Core 2.0.11 with IPv6 support, based on IDF v4.4.5.20230722](https://github.com/tasmota/platform-espressif32/releases/tag/2023.07.00) is available too.
Could you try that one as well? I'd rather upgrade to the newer version if that also fixes your issue.

I do own several D32 Lolin Pros, they are great devices! Any help is always appreciated, be it as code contributions, docs review, custom component etc. as well as donations. 🙂

jgoldwein, 2023-08-06:
Isn't that the release already built into the latest openHASP?  If so, I've tried that and the errors indeed do occur.  

JG

fvanroie, 2023-08-09:
Just released https://github.com/tasmota/platform-espressif32/releases/tag/2023.08.00

jgoldwein, 2023-08-15:
Unfortunately, using  https://github.com/tasmota/platform-espressif32/releases/tag/2023.08.00 does not solve the problem.

JG

jgoldwein, 2023-08-16:
After creating and testing multiple builds, I have now tracked down the problem to a change made between:

1. https://github.com/tasmota/platform-espressif32/releases/download/2023.05.02/platform-espressif32.zip
                                              and
2. https://github.com/tasmota/platform-espressif32/releases/download/2023.05.03/platform-espressif32.zip

i.e. https://github.com/tasmota/platform-espressif32/compare/2023.05.02...2023.05.03

Which possibly leads to this:

  https://github.com/tasmota/arduino-esp32/compare/2.0.9.20230529...2.0.9.20230531

For my build of openHASP for Lolin D32 Pro, the most recent openHASP build (openHASP 0.7.0-rc7) operates properly if the platform is changed from:
   platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.07.00/platform-espressif32.zip
to:
   platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.05.02/platform-espressif32.zip
 
I cannot see what the cause may be and think I have exhausted my ability to track down the source of this problem any further and appreciate your help.

Thanks

JG

marsman7, 2023-08-25:
I become the same error on an Lanbon L8 on version 0.7.0-rc7 with 

```
[env:lanbon_l8]
extends = arduino_esp32_v2, flash_8mb
board = esp32dev
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.07.00/platform-espressif32.zip
```

It runs without problems, but when I access the openHASP with a web browser the MQTT connection breaks, in the serial terminal I can see this happening:

```
[15:50:46.033][ 2932/ 6576 55][45100/45180  1] HTTP: Sending 200 /static/vars.css to client connected from: 192.168...
[15:50:46.094][ 2932/ 5888 50][45100/45180  1] HTTP: Sending 200 /static/style.css to client connected from: 192.168...
[15:50:46.135][ 1524/ 4168 63][45100/45180  1] HTTP: Sending 200 /static/main.js to client connected from: 192.168...
[15:51:02.491][ 1780/ 3876 54][45100/45180  1] MQTT: Getrennt
[15:51:21.516][ 2036/ 4252 52][45100/45180  1] MQTT: Transport error
[15:51:21.526][ 2036/ 4252 52][45100/45180  1] MQTT: Getrennt
[15:51:40.535][ 2036/ 5008 59][45100/45180  1] MQTT: Transport error
[15:51:40.545][ 2036/ 5008 59][45100/45180  1] MQTT: Getrennt
```
Sorry its german (getrennt = disconnect)

For me is this solved when i compile it with :

```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.08.00/platform-espressif32-2023.08.00.zip
```

The serial terminal output in the same situation 

```
[16:13:27.929][ 8692/12376 29][45100/45180  1] HTTP: Sending 200 /static/vars.css to client connected from: 192.168...
[16:13:27.982][ 8180/10320 20][45100/45180  1] HTTP: Sending 200 /static/style.css to client connected from: 192.168...
[16:13:28.027][11764/15812 25][45100/45180  1] HTTP: Sending 200 /static/main.js to client connected from: 192.168...
[16:13:28.081][ 4340/ 9952 56][45100/45180  1] HTTP: Sending 200 /static/petite-vue.hasp.js to client connected from: 192.168...
[16:13:28.128][15348/16012  4][45100/45180  1] HTTP: Sending 200 /static/en.json to client connected from: 192.168...
[16:13:28.174][11252/10360 11][45100/45180  1] HTTP: Sending 200 /static/logo.svg to client connected from: 192.168...
[16:13:38.679][14836/16668 10][45100/45180  1] MQTT PUB: p1b1 => {"event":"down","val":0}
[16:13:38.709][14836/16668 10][45100/45104  1] MQTT PUB: p1b1 => {"event":"up","val":1}
[16:13:38.723][14836/16420  9][45100/45104  1] MQTT PUB: output12 => {"state":"on"}
[16:14:38.764][14836/16668 10][45100/45180  1] MQTT PUB: idle => short
```

Now i see a differens. In the upper prints i remove the "Prompt >  " on begin of each line. On compile with platform 2023.07.00 is on left site of first "[" a additional character.

```
Prompt > �[15:50:46.033][ 2932/ 6576 55][45100/45180  1] HTTP: Sending ...
versus
Prompt >  [16:13:27.929][ 8692/12376 29][45100/45180  1] HTTP: Sending ...
```

I continue to test it

nagyrobi, 2023-08-25:
It's worth considering to drop the dependency on Tasmota on long term. They change core stuff pretty often which beaks things. That Tasmota-fork of the lib is intended solely for their own purposes, when they change stuff they don't think of backwards compatibility as they adapt their own project right away.
It's not good for OpenHASP to rely on such an unstable library. It's not the first time this happens.

fvanroie, 2023-08-25:
> It's not good for OpenHASP to rely on such an unstable library. It's not the first time this happens.

Well, Tasmota core is usually quicker to implement new stuff, it was chosen when the ESP32-S3 wasn't properly supported in Arduino 2.0 yet. It has served this project pretty well so far. Meanwhile there is 2023.08.01 out which might fix additional bugs.

Feel free to swap out the framework to your liking. If Arduino core is more stable, then we can consider it... Any library has it bugs and issues, so please test it and report back. Thanks

nagyrobi, 2023-08-26:
How about ESP-IDF? no Arduino wrapper at all? Less overhead, less memory consumption...

fvanroie, 2023-08-26:
Sure, if you are in favor of ESP-IDF, that's an option too. Can you test it?

nagyrobi, 2023-08-26:
A bit busy nowdays, we're reconstructing part of our house. Let's get back to it win 1-2 months.

marc-gist, 2023-08-29:
I had a similar issue that resulted in memory run out issues (i beilive), would disconnect from mqtt, then wifi restart, every time I tried to load up a 90 line pages.jsonl file (after enabling mqtt).

Roll back to `2023.05.01/platform-espressif32.zip`, which worked. 
Now testing `2023.08.01/platform-espressif32.zip` which seems to be working so far!

jgoldwein, 2023-08-30:
> I had a similar issue that resulted in memory run out issues (i believe), would disconnect from mqtt, then wifi restart, every time I tried to load up a 90 line pages.jsonl file (after enabling mqtt).
> 
> Roll back to `2023.05.01/platform-espressif32.zip`, which worked. Now testing `2023.08.01/platform-espressif32.zip` which seems to be working so far!

@marc-gist - Can you confirm that this is working?

fvanroie, 2023-09-27:
This seems to have been fixed in the newer Tasmota Core release. Can this issue be closed?

closed 2023-09-28 by fvanroie