HASwitchPlate/openHASP #515: MakerFabs 4.3 TFT support

hallard, 2023-08-06:
I got a Makerfabs ESP32 S3 Parallel like this one 
[image]

https://www.tindie.com/products/makerfabs/esp32-s3-parallel-tft-with-touch-43/

I already compiled and build my custom lvgldemo and platformio for this one and it works fine 
https://github.com/hallard/Makerfabs-LvglWidgetsDemo

So I wanted to give a try adding it to HASP and found there was already all files needed so I just added env for `makerfabs-s3-tft43-rgb` in my `platformio_override.ini`

Compiled and flashed, all works except lot of artifacts on display like sync issues. So I checked with my setup and changed 

```cpp
    -D TFT_HSYNC_POLARITY=0
    -D TFT_HSYNC_FRONT_PORCH=48   ; Maximum HSYNC Front Porch
    -D TFT_HSYNC_PULSE_WIDTH=4    ; Typical HSYNC Pulse Width
    -D TFT_HSYNC_BACK_PORCH=16    ; Typical HSYNC Back Porch
    -D TFT_VSYNC_POLARITY=0
    -D TFT_VSYNC_FRONT_PORCH=12   ; Maximum VSYNC Front Porch
    -D TFT_VSYNC_PULSE_WIDTH=4    ; Typical VSYNC Pulse Width
    -D TFT_VSYNC_BACK_PORCH=4    ; Typical VSYNC Back Porch
    -D TFT_PCLK_ACTIVE_NEG=1
    -D TFT_PREFER_SPEED=14000000  ; 1/2 of Typical DCLK Frequency
```

by my working setup just in case.

```cpp
    -D TFT_HSYNC_POLARITY=0
    -D TFT_HSYNC_FRONT_PORCH=48   ; Maximum HSYNC Front Porch
    -D TFT_HSYNC_PULSE_WIDTH=48   ; Typical HSYNC Pulse Width
    -D TFT_HSYNC_BACK_PORCH=88    ; Typical HSYNC Back Porch
    -D TFT_VSYNC_POLARITY=0
    -D TFT_VSYNC_FRONT_PORCH=13   ; Maximum VSYNC Front Porch
    -D TFT_VSYNC_PULSE_WIDTH=3    ; Typical VSYNC Pulse Width
    -D TFT_VSYNC_BACK_PORCH=32    ; Typical VSYNC Back Porch
    -D TFT_PCLK_ACTIVE_NEG=1
    -D TFT_PREFER_SPEED=16000000  ; 1/2 of Typical DCLK Frequency
```
But issue stil there.

I noticed web inteface very slow so network that may slow down display and may be the artifact issues.

[image]

I built from master config is as follow
```
openHASP
--
Version | 0.7.0-rc7 19ed334
Build DateTime | Aug 6 2023 23:11:45 UTC
Environment | makerfabs-s3-tft43-rgb
Uptime | 6m 41s
Idle | long
Active Page | 1

Free Heap | 89.95 KiB
Free Block | 75.98 KiB
Fragmentation | 15%
PSRam Free | 7.25 MiB
PSRam Size | 7.99 MiB

Total | 64.00 KiB
Free | 57.09 KiB
Fragmentation | 1%

Model | ESP32-S3 rev0
Frequency | 240MHz
Core Version | 4.4.5.230722
Reset Reason | CPU0: POWERON_RESET / CPU1: POWERON_RESET
Flash Size | 16.00 MiB
Program Size Used | 1.57 MiB
Program Size Free | 1.81 MiB
Filesystem Size | 320.00 KiB
Filesystem Used | 8.00 KiB
Filesystem Free | 312.00 KiB
```

Any idea why this issue so I can do some tests to check?

We're so close to get it working, worth fixit it :-)

hallard, 2023-08-17:
Received today 2 Sunton, one 5" and one 7" and add exactly the same issue

After lot on investigation I got it all (including 4.3 one) working, 

The deal is the incorrect flash mode for sunton (may be other) it SHOULD be dio so added this mode into file `sunton-esp32-s3-tft.ini` as follow (may be it's not the best place but now it works).

```ini
board_build.flash_mode = dio
```

looks like that after change

```ini
;***************************************************;
;    Sunton ESP32-S3 TFT 4.3", 5.0", 7.0"           ;
;          - Custom esp32-s3 board                  ;
;          - ili6485 480x272 or st7262 IPS 800x480  ;
;          - xpt2046 or gt911 touch controller      ;
;***************************************************;

[sunton-esp32-s3-tft]
extends = arduino_esp32s3_v2
board = esp32-s3-devkitc-1
board_build.arduino.memory_type = qio_opi
board_build.flash_mode = dio

```

fvanroie, 2023-08-24:
Is this issue about the MakerFabs 4.3 TFT support or the Sunton S3?

hallard, 2023-08-24:
Solved both sunton and makerfabs on my side