HASwitchPlate/openHASP #520: Anti-burn-in does not respect plate orientation

akloeckner, 2023-08-19:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

When running anti-burn-in on my 90° rotated Lanbon L8, a slim area on the right side of the display is not trained. I suppose, the anti-burn-in uses the default screen size (240x320), but the display is rotated (320x240).

### To Reproduce

I am not sure, how to do this with code.

### Expected behavior

The anti-burn-in should cover the entire screen. Either the size of the random pixels should be adapted to the screen rotation. Or, maybe easier, just a square should be generated (using the larger dimension).

### Screenshots or video

fvanroie, 2023-08-19:
Please provide the firmware version from the information page.

akloeckner, 2023-08-19:
I am on 0.6.3.

fvanroie, 2023-08-19:
This issue should already be fixed in 0.7.0-rc #389

akloeckner, 2023-08-19:
:+1: Thanks. I will check back, when the release is official. If this is still an issue then, I'll reopen this issue.

closed 2023-08-19 by akloeckner