HASwitchPlate/openHASP #528: Makerfabs S3 tft35 spi display issues

jonofmac, 2023-08-24:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug
The display has lots of white dots after the initial boot image.
The hardware is explicitly called out as supported in the documentation (at least for version 0.7.0), but I realize this is early release.
https://www.openhasp.com/0.7.0/hardware/makerfabs/esp32-s3-spi-tft-touch/

What is interesting is sometimes when I tap the OK button, the part it re-draws is without the white dots.

### To Reproduce
I am building from source 0.7.0-rc6 and using the makerfabs-s3-tft35-spi env. No changes.

### Expected behavior
Display without dots.

### Steps I've tried

1. I uploaded the default makerfabs demo code, display is clear
2. I've modified the env file to slow down SPI frequency to the display. No change in dots, just slower drawing.

### Screenshots or video
[image]
[image]

Screen shot from the web interface shows no artifacts.

fvanroie, 2023-08-24:
I think this config needs to be added to the `user_setups\esp32s3\makerfabs-s3-tft35-spi.ini` config:
```ini
    -D TFT_DMA_CHANNEL=1
```
I haven't verified it, but if you can compile it in and test that would be great.

jonofmac, 2023-08-24:
Confirmed that adding `-D TFT_DMA_CHANNEL=1` to the build_flags fixed the visual issues. Thank you.
Found another issue with the .ini file under build_flags. The GPIO for TFT backlight is set to -1, it should be ` -D TFT_BCKL=48`

Once I added GPIO48 as the TFT backlight, then dimming works as expected, as well.

fvanroie, 2023-08-24:
Thanks for your report and testing. I have added both flags to the config, so it should be OK now.
I'll close this issue, feel free to post an update if you find any more problems.

closed 2023-08-24 by fvanroie