HASwitchPlate/openHASP #534: D1 R32 ILI9486 Failed to start: src/drv/tft/tft_driver_lovyangfx.cpp

dfsx1, 2023-09-05:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version 
       (I tried v0.7.0 but that version is not working at all for me, I get only gibberish console output)
- [x] Reproduce the issue and describe all steps

### Describe the bug

>
[00:38:29]        Home Automation Switch Plate
[00:38:29]        Open Hardware edition v0.6.3
[00:38:29]
[00:38:29][       0.053][110580/184484 40][    0/    0  0] DBUG: Started @ 115200 bps
[00:38:29]#[       0.065][110580/184484 40][    0/    0  0] DBUG: Environment: d1-r32-unoshield_ili9486_adc
[00:38:29]#[       0.078][110580/184484 40][    0/    0  0] UART: Started
[00:38:29]#[       0.089][110580/184484 40][    0/    0  0] UART: Client login from serial
[00:38:29]#[       0.101][110580/184484 40][    0/    0  0] CONF: Failed to load /config.json
[00:38:29]Prompt > [       0.114][110580/184484 40][    0/    0  0] CONF: Loading EEPROM
[00:38:29]Prompt > [       0.126][110580/184484 40][    0/    0  0] CONF: Loaded EEPROM
[00:38:29]Prompt > [       0.138][110580/184544 40][    0/    0  0] DBUG: Loading debug settings
[00:38:29]Prompt > [       0.151][110580/184544 40][    0/    0  0] GPIO: Loading GUI settings
[00:38:29]Prompt > [       0.164][110580/184544 40][    0/    0  0] HASP: Loading HASP settings
[00:38:29]Prompt > [       0.177][110580/184544 40][    0/    0  0] WIFI: Loading WiFi settings
[00:38:29]Prompt > [       0.190][110580/184544 40][    0/    0  0] MQTT: Loading MQTT settings
[00:38:29]Prompt > [       0.203][110580/184544 40][    0/    0  0] TELN: Loading Telnet settings
[00:38:29]Prompt > [       0.216][110580/184544 40][    0/    0  0] MDNS: Loading MDNS settings
[00:38:29]Prompt > [       0.228][110580/184544 40][    0/    0  0] HTTP: Loading HTTP settings
[00:38:29]Prompt > [       0.241][110580/184544 40][    0/    0  0] GPIO: Loading GPIO settings
[00:38:29]Prompt > [       0.254][110580/184544 40][    0/    0  0] CONF: Settings loaded
[00:38:29]Prompt > [       0.266][110580/186092 40][    0/    0  0] MSGR: Starting...
[00:38:29]Prompt > [       0.278][110580/186092 40][    0/    0  0] MSGR: Started
[00:38:29]Prompt > [       0.289][110580/186092 40][    0/    0  0] TFT : Starting...
[00:38:29]Prompt > [       0.301][110580/186092 40][    0/    0  0] TFT : Starting...
[00:38:29]Prompt > [       0.313][110580/185416 40][    0/    0  0] TFT : Failed to start: src/drv/tft/tft_driver_lovyangfx.cpp line 295

### To Reproduce
Run binaries from the webinstallers for d1-r32-unoshield_ili9486_adc

### Expected behavior
When I look at the v0.6.3 code I only see ILI9488 inside the `switch(tft_driver){}`
ILI9486 is missing there?
https://github.com/HASwitchPlate/openHASP/blob/497f611b03537dd69bd1386fc3fe9e1f8fc39c60/src/drv/tft/tft_driver_lovyangfx.cpp#L257-L297

Even in the master branch (v0.7.0) there is nothing for ILI9486
https://github.com/HASwitchPlate/openHASP/blob/e2910604a2ce4d979c47fea352b2d586b5b9ca64/src/drv/tft/tft_driver_lovyangfx.cpp#L284-L328

I see that `lgfx::Panel_ILI9486()` library actually exists and is different from `lgfx::Panel_ILI9488()` so I tried just adding it to the code.
```
case 0x9486: {
	panel = new lgfx::Panel_ILI9486();
	LOG_DEBUG(TAG_TFT, F("%s - %d"), __FILE__, __LINE__);
	break;
}
```
Unfortunatelly I run into generic compile errors need more time to setup the build environment.

Can someone clarify me if I'm (strange enough) the first person testing this ILI9486 while binaries are available for years. 
Or could I be overlooking something else and searching in the wrong direction?

fvanroie, 2023-09-13:
IIRC the analog touch driver was code contributed by a user. I don't think there are many users with this hardware combination out there. As such, the combination is tested (often) and relies an user feedback. I'll have a look...

It's always best to start with the display config first and see that those pins are correct.
Touch pins can be set to any of the free pins as to not interfere with the display.

dfsx1, 2023-10-01:
@fvanroie
After some research I learned ESP32 pins 34,35,36,39 are input pins only.
Hardware modification was needed for R1 D32 as documented here: https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support

[image]

I see a screenshot in the analog touch driver contributors pull request #213
Don't understand how @wesleygas used pin 34 for TFT_WR as this is already connected to TFT_CS on the shield.
I chose to use pin 39 instead of 34 for TOUCH_anWR, to keep pin 33=>34 compatible with the mod described in TFT_eSPI docs.

[image]

After the hardware mod (and fix #553 for this error msg) I got the display working.
But with LovyanGFX I only see half of some kind of bootscreen.
Pages don't update and no touch.
(TFT : Touch Failed to start).

[image]

TFT_eSPI works fine, only no touch.
(#warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp])
I'm wondering if I can get the analog touch driver working for TFT_eSPI 🤔

[image]

commit 1b244786 (2023-10-01): Merge pull request #553 from dfsx1/master

Fix #534 ILI9486 Failed to start: src/drv/tft/tft_driver_lovyangfx.cpp
commit ad0f69f6 (2023-10-01): Fix #534 ILI9486 Failed to start: src/drv/tft/tft_driver_lovyangfx.cpp
closed 2023-10-01 by fvanroie