HASwitchPlate/openHASP #540: Panel-Hosted file editor is non-functional

fake-name, 2023-09-13:
### Perform all steps below and tick them with [x]
- [ x ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [ x ] Update openHASP to the latest version
- [ x ] Reproduce the issue and describe all steps

### Describe the bug

The file editor hosted on the panel (e.g. at `<panel_ip>/edit` does not work. 

Looking closer, it appears to depend on externally hosted javascript file (Cloudflare! Really?)

### To Reproduce

Go to file editor. 

Expect to be able to actually edit files.

### Expected behavior

I would expect a device which is part of a offline-first home-automation system to actually be offline. 

My whole home-automation system is on a walled off VLAN with no internet access.

fvanroie, 2023-09-13:
There is one dependency on the `ace.js` script for the File Editor, yes. If you wish, you can host the library internally on your network and change all references in the firmware.

It's common practice to use CDNs to deliver static web content. The javascript library is run within your browser.
Is your PC also in the restricted VLAN? Or did you disable JavaScript entirely?
What specifically makes this setup a security issue? What's wrong with Cloudflare?

The device can work fine without the File Editor functionality, just use curl HTTP requests or FTP to upload files or send pages files via MQTT.

> I would expect a device which is part of a offline-first home-automation system to actually be offline.

The device itself *is* offline, just part of the File Editor assumes your **PC** has an internet connection to get the required editor to edit the files. I don't really see this as being unreasonable or particularly unsafe.

Because of the size of ace.js it seemed unreasonable to include the whole library in the firmware, just to edit a text file.

fake-name, 2023-09-14:
> It's common practice to use CDNs to deliver static web content. 

There is zero internet web content on the device. 

It's a security issue because there is zero guarantee that the same contents of `ace.js` is served to everyone, every time it's requested. Unless you're using a content security policy with a hash of the related js file, there is zero protection against a malicious CDN, the library itself being taken over by a malicious actor, or things just plain-old breaking. 

> The device itself is offline, just part of the File Editor assumes your PC has an internet connection to get the required editor to edit the files. I don't really see this as being unreasonable or particularly unsafe.

So the device is offline except for all the places it isn't? Why is any content hosted on the device at all, aside from an `index.html` in that case?

fvanroie, 2023-09-14:
> There is zero internet web content on the device.

Correct.

> So the device is offline except for all the places it isn't?

The device doesn't connect to the internet, your PC does. Chrome, Firefox, Edge nor Safari work fine with this setup.

Yes, a CDN could host bad scripts, which *any* web server can do. Like I said, you can avoid this by hosting the files locally.

If your requirement is a strict content-security-policy for every file on the internet, I'm sure there are ways to implement this too.
If you have the expertise, please provide possible solutions or suitable Content Security Policy (CSP).

> Why is any content hosted on the device at all, aside from an index.html in that case?

It's for user-convenience. You can opt to disable the Web Service altogether if so inclined, which some users do.
As I explained, everything is hosted on the device, *except* the `ace.js` of the File Editor which is due a size constraint.

fvanroie, 2024-02-02:
Closed, without feedback.

closed 2024-02-02 by fvanroie