HASwitchPlate/openHASP #544: Scrambled display on ESP32-2432S028

rodavrow, 2023-09-15:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x ] Update openHASP to the latest version
- [x ] Reproduce the issue and describe all steps

### Describe the bug
I flashed the ESP32-2432S028 using the web tool with esp32-2432s028r_full_4MB_v0.7.0-rc6_2f4092b.bin - issue is as seen with the screen being misaligned, mirrored and unresponsive to touch.

I then flashed using the latest Github actions nightly to esp32-2432s028r_full_4MB_v0.7.0-rc8_f28627c.bin - issue is the same.

Console operates as expected.

### To Reproduce

Flash the ESP32-2432S028 Sunton board with any version of the firmware.

### Expected behavior
The screen should display and operate normally.

### Screenshots or video

[image]
[image]

rodavrow, 2023-09-15:
OK, I may have found the issue. I tried loading ESP-Home using some example code for this board, and I believe this new revision with the USB-C Controller may have a different display controller.

It appears it used to be ILI9341 based on the code I used, but when I loaded the firmware I had exactly the same display issue. I then changed the configration in ESP-Home to ILI9342 it now works as expected.

I will attempt to see what I need to change in openHASP and test.

rodavrow, 2023-09-18:
I'm a bit lost on how to implement the ILI9342 driver, can anyone assist please?

FreezyExp, 2023-10-25:
just set width as height and rotate the screen see
https://github.com/HASwitchPlate/openHASP/issues/550#issuecomment-1737523475

rodavrow, 2023-10-25:
Thanks! I felt that the colors were wrong though. Not sure if relevant to this as referenced here:

https://forum.arduino.cc/t/i-have-problems-to-use-my-tft-screen-2-4-ili9342-help-me-please/455398

> I tryed a lot of libraries for ILI9341, none works, except maybe one.

> I say maybe because :

> colors are inverted, black is 0xFFFF and wite 0X0000 , rgb565.
> texte is miroired and biggining from right side
> portrait screen is displayed and splited in lanscape screen or lanscape screen is diplayed and splited in portrait screen

yeyeto2788, 2023-11-03:
The colors are also inverted, BGR instead of RGB I faced the same issue yesterday trying to adapt the board and make it work.

https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/issues/40

Grey-Lancaster, 2024-01-18:
+1
needs to be a dropdown on the web install page for the ili9342 as ili9341 does not work

yeyeto2788, 2024-01-22:
Hey @Grey-Lancaster I got it to work with the following code. You still need to configure screen rotation and mirroring through WiFi AP.

Here you have the binary compile in case you're interested.
[esp32-2432s028r_full_4MB_v0.7.0-rc9_c200edf.bin.zip](https://github.com/HASwitchPlate/openHASP/files/14006612/esp32-2432s028r_full_4MB_v0.7.0-rc9_c200edf.bin.zip)

I had to modify a configuration file in order to get it working `user_setups/esp32/esp32-2832s028.ini`

```ini
;***************************************************;
;       Sunton ESP32-WROOM custom dev board with    ;
;               - ILI9341 TFT SPI 4-WIRE            ;
;               - XPT2046 touch controller          ;
;***************************************************;

[esp32-2432s028r]
extends = arduino_esp32_v2
board = esp32dev
monitor_port = COM4       ; Change to the correct port
upload_port = COM4
upload_protocol = esptool
upload_flags =
    --erase-all

build_flags =
    ${arduino_esp32_v2.build_flags}
    ${esp32.no_ps_ram}

;region -- TFT_eSPI build options ------------------------
    ; -D USER_SETUP_LOADED=1
    -D LGFX_USE_V1=1
    -D ILI9342_DRIVER=1
    -D INVERT_COLORS=1
    -D TFT_ROTATION=0 ; 0=0, 1=90, 2=180 or 3=270 degree
    -D TFT_WIDTH=320
    -D TFT_HEIGHT=240
    -D TFT_RST=-1
    -D TFT_SCLK=14
    -D TFT_DC=2
    -D TFT_CS=15
    -D TFT_MOSI=13
    -D TFT_MISO=12
    -D TFT_BCKL=21
    -D SUPPORT_TRANSACTIONS
    -D SPI_FREQUENCY=55000000
    -D SPI_READ_FREQUENCY=20000000
    -D TOUCH_OFFSET_ROTATION=0  ; 1=swap xy, 2=invert x, 4=inverty

    -D ESP32_2432S028R=1
;endregion

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG

lib_deps =
    ${arduino_esp32_v2.lib_deps}
    ${lovyangfx.lib_deps}

[env:esp32-2432s028r_4MB]
extends = esp32-2432s028r, flash_4mb
build_flags =
    ${esp32-2432s028r.build_flags}
    -D TOUCH_DRIVER=0x2046 ; XPT2606 Resistive touch panel driver
    -D HASP_USE_LGFX_TOUCH=1
    -D TOUCH_CS=33
    -D TOUCH_SCLK=25
    -D TOUCH_MOSI=32
    -D TOUCH_MISO=39
    -D TOUCH_SDA=-1
    -D TOUCH_SCL=-1
    -D TOUCH_IRQ=36
    -D SPI_TOUCH_FREQUENCY=2500000

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE

lib_deps =
    ${esp32-2432s028r.lib_deps}
    ;${tft_espi.lib_deps}
    ${lovyangfx.lib_deps}
    ;${goodix.lib_deps}
```

Grey-Lancaster, 2024-01-22:
No clue, colors are terrible and touch is not working.  And for some reason HA is not discovering it.  Added the plate to yaml, uploaded pages.jason.  Info says connected to mqtt

oklona, 2024-01-26:
Hmm... I just applied the fixes from ILI9341_2 as suggested by #550. 

In plain English, I found the settings (lots of writecommands) here: [https://github.com/Bodmer/TFT_eSPI/issues/1172], and applied them to the init-function in lib/GxTFT/src/GxCTRL/GxCTRL_ILI9341/GxCTRL_ILI9341.cpp. Now my display (with USB-C, which seems to be the sign of the new revision of the board) actually comes up correctly after a regular rebuild/reflash!

So, this fix should probably be added to the official code. However, I am unsure what would be the best approach. I have seen some references to this TFT chip as ILI9342 instead of 9342, so the question is whether it should be implemented as a new driver (hugely based on the ILI9342 but with the init-function writing different values), or if the init function should just be updated for ILI9341. According to the TFT_eSPI post, it seems the changes should work for all displays, but obviously that should be tested and verified.

fvanroie, 2024-02-27:
I added a new ili9342 firmware, but can't test it myself. You can get it from [the latest build](https://github.com/HASwitchPlate/openHASP/actions/runs/8072458225) in the [Actions tab](https://github.com/HASwitchPlate/openHASP/actions) for testing. 

Please let me know the result.

zalves, 2024-02-27:
I've installed the ili9342 firmware and it still get part of the display scrambled, colors inverted. 

[image]

zalves, 2024-02-27:
The box does come with a sticker saying 7789 but I've tried the firmware with st7789 and got a messy result also. [image]

[image]

Only thing that worked visually was @yeyeto2788 build, but it doesn't get discovered by home assistant.

fvanroie, 2024-02-28:
I swapped the width and height like @FreezyExp  suggested, feel free to test again.

zalves, 2024-02-28:
I've tested, last build and works fine :D configure display to 90º mirrored.

Thank you @fvanroie

smirnowegor, 2024-05-29:
> I've tested, last build and display works fine :D configure display to 90º mirrored, calibrated touch by pressing opposite to target in calibration, otherwise it's inverted
> 
> Thank you @fvanroie

Hi! Did you calibrate it via GUI? 
when i calibrate device it works fine ... until device is rebooted. Than i have to calibrate it again

Vad1mka2022, 2024-12-19:
@fvanroie  maybe i don't understand how it works but i can't find a fixed firmware
the links you gave are no longer active and there is no firmware file there. There is no choice here either https://nightly.openhasp.com/

zalves, 2024-12-19:
> @fvanroie maybe i don't understand how it works but i can't find a fixed firmware the links you gave are no longer active and there is no firmware file there. There is no choice here either https://nightly.openhasp.com/

I have a firmware that works with a small caveat, the RGB order doesn't seem to be correct. so if you need a color picker it will be off. I can provide it if you wish

fvanroie, 2024-12-19:
You can always use the latest build action [artifacts here](https://github.com/HASwitchPlate/openHASP/actions/workflows/build.yaml). There is an ILI9342 firmware for this device in the Sunton.zip file. Please report if it works or if there are issues with it.

As noted, I'm reliant on feedback as I can't test it locally...

Vad1mka2022, 2024-12-19:
I checked, everything works as for the screen. I used this file
esp32-2432s028r-ili9342_full_4MB_v0.7.0-rc15_7e5762d.bin . I still need to
check the integration with Home Assistant mqtt, maybe I'm doing something
wrong or the latest firmware version is needed. You can add support for
this screen to the final version. Thank you for your work

stamanf, 2025-01-27:
Total noob here, how do you flash the bin file ?

c4ve-d, 2025-02-10:
Hi

I have the same display running openHASP.

Same display, same packaging.

I did some testing with a spare one:

_**"Original" Firmware:**_ (esp32-2432s028r_full_4MB_v0.7.0-rc13_042fe05)
Part of the display scrambled, as shown in pictures above.

_**ILI9342 firmware:**_ (esp32-2432s028r-ili9342_full_4MB_v0.7.0-rc13_042fe05)
This firmware des not run with a correct Display output out of the box.
-Dispaly with this firmware is color inverted. -> disable checkbox "Invert Colors" solves the problem
-Calibration of the touchscreen necessary.
-So far it is running... If I find problems, I will report them

```
display_test
openHASP
--
Version | 0.7.0-rc13 042fe05
Build DateTime | Aug 26 2024 00:44:21 UTC
Environment | esp32-2432s028r-ili9342_4MB
Uptime | 3m 49s
Idle | short
Active Page | 1

Free Heap | 72.95 KiB
Free Block | 67.98 KiB
Fragmentation | 6%

Total | 48.00 KiB
Free | 41.48 KiB
Fragmentation | 1%

Server
Username
Client ID
Status | Disabled
Received
Published
Failed

BSSID | 26:00:00:00:00:00
SSID | Corporation
Signal Strength | -60dBm (Fair)
IP Address | 10.0.20.60
Gateway | 10.0.20.1
DNS Server | 10.0.20.1
MAC Address | 88:13:00:00:00:00

Model | ESP32-D0WD-V3 rev3
Frequency | 240MHz
Core Version | 4.4.6
Reset Reason | CPU0: SW_CPU_RESET / CPU1: SW_CPU_RESET
Flash Size | 4.00 MiB
Program Size Used | 1.63 MiB
Program Size Free | 1.68 MiB
Filesystem Size | 560.00 KiB
Filesystem Used | 12.00 KiB
Filesystem Free | 548.00 KiB
```

My "old" running firmware:

Unfortunately, I don't remember exactly where I got the software image.
esp32-2432s028r-st7789_full_4MB_v0.7.0-rc12_ca346e8.bin

Here the info data:

```
display_1
openHASP
Version	0.7.0-rc12 ca346e8
Build DateTime	Aug 25 2024 21:34:18 UTC
Environment	esp32-2432s028r-st7789_4MB
Uptime	39s
Idle	off
Active Page	1
Device Memory
Free Heap	67.96 KiB
Free Block	63.98 KiB
Fragmentation	5%
LVGL Memory
Total	48.00 KiB
Free	35.69 KiB
Fragmentation	1%
MQTT
Server
Username
Client ID
Status	Disabled
Received
Published
Failed
Wifi
BSSID	26:00:00:00:00:00
SSID	Corporation
Signal Strength	-59dBm (Good)
IP Address	10.0.20.59
Gateway	10.0.20.1
DNS Server	10.0.20.1
MAC Address	CC:7B:00:00:00:00
Module
Model	ESP32-D0WD-V3 rev3
Frequency	240MHz
Core Version	4.4.6
Reset Reason	CPU0: POWERON_RESET / CPU1: EXT_CPU_RESET
Flash Size	4.00 MiB
Program Size Used	1.63 MiB
Program Size Free	1.68 MiB
Filesystem Size	560.00 KiB
Filesystem Used	16.00 KiB
Filesystem Free	544.00 KiB
```

aamarques, 2025-08-14:
> You can always use the latest build action [artifacts here](https://github.com/HASwitchPlate/openHASP/actions/workflows/build.yaml). There is an ILI9342 firmware for this device in the Sunton.zip [...]

Hey.. I'm late :)

I have installed the firmware from @yeyeto2788 and works great (The colors not match - I don't care). 
So, following your recomendation, I downloded the latest Suton.zip and installed this ` esp32-2432s028r-ili9342_ota_v0.7.0.1_acebfb9.bin`, and the layout are ok but the touch are inverted. I mean, If you have a button near the botton, the touch is in the upper. If the button is on left, the touch is on right.

fvanroie, 2025-11-16:
Recalibrate the Display touchpanel via the Web UI and it should work fine!

closed 2025-11-16 by fvanroie