HASwitchPlate/openHASP #550: ILI9341_2_DRIVER support

FreezyExp, 2023-09-27:
**Is your feature request related to a problem? Please describe.**
When i build for my esp32-2432s028, using the esp32-2432s028r configuration, the screen is mirrored.
Also I need to set the height as the width to get it to render the correctly.

**Describe the solution you'd like**
According to the manual that came with the device, it requires the User_Setup.h of the TFT_eSPI library to declare ILI9341_2_DRIVER instead of ILI9341_DRIVER 
but ILI9341_DRIVER is used in various locations throughout the project.

**Describe alternatives you've considered**
I will look into mirroring the output, that will get me one step closer without needing modifications.
rotation 6 did not seem to mirror it, but I will update this request when I have investigated further.

Since TFT_eSPI is not used in openHASP (outside of testing?), perhaps investigating what they did differently with that pragma could fix the display.

**Additional context**

FreezyExp, 2023-09-27:
Setting rotation 6 seems to have had no effect at all.

fvanroie, 2023-09-27:
Does the board have a USB-C type connector or MicroUSB?

It seems they release a [new revision of the board with USB-C ](https://github.com/HASwitchPlate/openHASP/issues/544)and a different display controller.

FreezyExp, 2023-09-27:
with USB-C, after building with rotate 6 the image was still flipped.
Rotating it back to 0 in configuration and restarting seems to have fixed the display, but now the touch input is mirrored in both x and y.

I removed the rotation and flashed again.
Changing the display configuration + restart I can easily get the screen to correct by selecting either or the mirrored options.

The initial TFT_ROTATION can be left at 0 and then use the http site to set it to 0 degrees mirrored.

To get the screen to at least render to the correct pixels I just did:
    -D TFT_WIDTH=320
    -D TFT_HEIGHT=240
 
 This still leaves the touch screen flipped in both X and Y, but that can probably be configured somewhere.

FreezyExp, 2023-09-27:
p.s. I am new to this repository and figuring this out as I go XD

My initial read of the user-manual pointed me here:
#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172

Hence my assumption that this might or might not be related / help.

Here is some data from the info section, just in case it might be useful.

<html>
<body>

Module
--
Model | ESP32-D0WD-V3 rev3
Frequency | 240MHz
Core Version | 4.4.5.230820
Reset Reason | CPU0: SW_CPU_RESET / CPU1: SW_CPU_RESET
Flash Size | 4.00 MiB
Program Size Used | 1.62 MiB
Program Size Free | 1.68 MiB
Filesystem Size | 560.00 KiB
Filesystem Used | 436.00 KiB
Filesystem Free | 124.00 KiB

</body>
</html>

FreezyExp, 2023-09-27:
Managed to get it working:
1. set rotation for lcd to 4 and touch to 1
2. build and flash
3. connect to device (QR code is ok even if it's mirrored)
4. configure display to 0 degrees
5. restart
6. calibrate display

Display and Touch are now working.
It's a bit of a mess, but it's ok.

```
;***************************************************;
;       Sunton ESP32-WROOM custom dev board with    ;
;               - ILI9341 TFT SPI 4-WIRE            ;
;               - XPT2046 touch controller          ;
;***************************************************;

[esp32-2432s028]
extends = arduino_esp32_v2
board = esp32dev
upload_speed = 921600

build_flags =
    ${arduino_esp32_v2.build_flags}
    ${esp32.no_ps_ram}

; TFT_ROTATION values:
; 0 - 0 deg (= default when not set)
; 1 - 90 deg anti-clockwise (from 0 deg)
; 2 - 180 deg anti-clockwise
; 3 - 270 deg anti-clockwise
; 4 - mirror content, and rotate 180 deg anti-clockwise
; 5 - mirror content, and rotate 270 deg anti-clockwise
; 6 - mirror content, and rotate 0 deg anti-clockwise
; 7 - mirror content, and rotate 90 deg anti-clockwise

;region -- TFT_eSPI build options ------------------------
    ; -D USER_SETUP_LOADED=1
    -D LGFX_USE_V1=1
    -D ILI9341_DRIVER=1 ;
    -D TFT_ROTATION=4 ; 0=0, 1=90, 2=180 or 3=270 degree, Mirrors: 6, 5, 4, 3
    -D TFT_WIDTH=320
    -D TFT_HEIGHT=240
    -D TFT_RST=-1
    -D TFT_SCLK=14
    -D TFT_DC=2
    -D TFT_CS=15
    -D TFT_MOSI=13
    -D TFT_MISO=12
    -D TFT_BCKL=21
    -D SUPPORT_TRANSACTIONS
    -D SPI_FREQUENCY=55000000
    -D SPI_READ_FREQUENCY=20000000
    -D TOUCH_OFFSET_ROTATION=1  ; 1=swap xy, 2=invert x, 4=inverty

    -D ESP32_2432S028R=1
;endregion

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG

lib_deps =
    ${arduino_esp32_v2.lib_deps}
    ${lovyangfx.lib_deps}

[env:esp32-2432s028_4MB]
extends = esp32-2432s028, flash_4mb
build_flags =
    ${esp32-2432s028.build_flags}
    -D TOUCH_DRIVER=0x2046 ; XPT2606 Resistive touch panel driver
    -D HASP_USE_LGFX_TOUCH=1
    -D TOUCH_CS=33
    -D TOUCH_SCLK=25
    -D TOUCH_MOSI=32
    -D TOUCH_MISO=39
    -D TOUCH_SDA=-1
    -D TOUCH_SCL=-1
    -D TOUCH_IRQ=36
    -D SPI_TOUCH_FREQUENCY=2500000

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE

lib_deps =
    ${esp32-2432s028.lib_deps}
    ;${tft_espi.lib_deps}
    ${lovyangfx.lib_deps}
    ;${goodix.lib_deps}
```

FreezyExp, 2023-09-27:
It is still weird to me that the device requires the mirror to be set and then disabled, but I can live with this setup step.

closed 2023-09-27 by FreezyExp