HASwitchPlate/openHASP #551: petite-vue-i18n-lite.1.0.2.min.js.gz wont go into filesystem

FreezyExp, 2023-09-27:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://www.openhasp.com/)
- [X] Update openHASP to the latest version
- [X] Reproduce the issue and describe all steps

### Describe the bug
unable to open '/static/petite-vue-i18n-lite.1.0.2.min.js.gz.
error adding file!
Error for adding content from static!

### To Reproduce
In platform io on windows, try to build the filesystem.
Also uploading to device in file editor and ftp fails.

### Expected behavior
files meant to be in data should be able to exist on the data partition

I think the filename might be too long or complicated.
I redownloaded the file and inspected it locally, initially it was corrupted (how git clone managed to get it corrupted without recognizing it as a file change is beyond me).

fvanroie, 2023-09-27:
There is no need to build the Filesystem image. So, this is not a real issue.
The data files are included into the firmware file and the filesystem is formatted at first boot.
No filesystem needs to be uploaded.

FreezyExp, 2023-09-27:
Ah ok, so because I had a corrupted file, it never made it to the device and I should erase and start over.