HASwitchPlate/openHASP #555: ESP32-2432s024c feature request

elik745i, 2023-10-09:
**Is your feature request related to a problem? Please describe.**
ESP32-2432s024c has ILI9341 TFT and CST820 capacitive touch screen. 

**Describe the solution you'd like**
Need support for CST820 Capacitive Touchscreen used in ESP32-2432s024c boards

**Describe alternatives you've considered**
ESP32-2432s024 boards has two variations r - with resistive and c - with capacitive touchscreen. Both would be handy to have in openHASP

**Additional context**
I tried to integrate arduino libraries into openHASP - master, but couldn't understand the principle of .pıo file generation, would be good if developers would write detailed instructions on how to integrate any arduino library into openHASP along with CST820 capacitive touch support.

elik745i, 2023-10-09:
I have figured it out, dont know write way or not but touch working now.

What needs to be done:
1) Create file in user_setups folder, name it esp32-2432s024.ini
`;***************************************************;
;       Sunton ESP32-WROOM custom dev board with    ;
;               - ILI9341 TFT SPI 4-WIRE             ;
;               - XPT2046 or CST820 touch controller ;
;***************************************************;

[esp32-2432s024]
extends = arduino_esp32_v2
board = esp32dev

build_flags =
    ${arduino_esp32_v2.build_flags}
    ${esp32.no_ps_ram}

;region -- TFT_eSPI build options ------------------------
    ${esp32.hspi}        ; Use HSPI hardware SPI bus
    -D ILI9341_DRIVER=1
    -D TFT_ROTATION=0 ; 0=0, 1=90, 2=180 or 3=270 degree
    -D TFT_WIDTH=240
    -D TFT_HEIGHT=320
    -D TFT_CS=15      ;// Chip select control pin
    -D TFT_DC=2       ;// Data Command control pin
    -D TFT_RST=-1     ;// Reset pin (could connect to RST pin)
    -D TFT_BCKL=27    ;None, configurable via web UI (e.g. 2 for D4)
    -D SUPPORT_TRANSACTIONS
    -D SPI_FREQUENCY=65000000
    -D SPI_READ_FREQUENCY=20000000
;endregion

; -- Debugging options -----------------------------
;   -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG

[env:esp32-2432s024r_4MB]
extends = esp32-2432s024, flash_4mb
build_flags =
    ${esp32-2432s024.build_flags}
    -D USER_SETUP_LOADED=1
    -D TOUCH_CS=33
    -D TOUCH_DRIVER=0x2046 ; XPT2606 Resistive touch panel driver
    -D HASP_USE_LGFX_TOUCH=1
    -D TOUCH_SDA=-1
    -D TOUCH_SCL=-1
    -D TOUCH_IRQ=-1
    -D I2C_TOUCH_PORT=0
    -D I2C_TOUCH_ADDRESS=0
    -D I2C_TOUCH_FREQUENCY=0
    -D SPI_TOUCH_FREQUENCY=2500000

lib_deps =
    ${esp32-2432s024.lib_deps}
    ${tft_espi.lib_deps}
    ;${lovyangfx.lib_deps}
    

[env:esp32-2432s024c_4MB]
extends = esp32-2432s024, flash_4mb
board = esp32dev

build_flags =
    ${esp32-2432s024.build_flags}
    ${esp32.no_ps_ram}

;region -- TFT_eSPI build options ------------------------
    ${esp32.hspi}        ; Use HSPI hardware SPI bus
    ;-D USER_SETUP_LOADED=1
    -D LGFX_USE_V1=1
[... 1353 lines cut ...]
    return TFT_PANEL_RM68140;
#elif defined(EPD_DRIVER)
    return TFT_PANEL_EPD;
#elif defined(RGB_DRIVER)
    return TFT_PANEL_RGB;
#else
    return TFT_PANEL_UNKNOWN;
#endif
}

uint32_t LovyanGfx::get_touch_driver()
{
#ifdef TOUCH_DRIVER
    return TOUCH_DRIVER > 0 ? TOUCH_DRIVER : 0;
#else
    return 0;
#endif
}

} // namespace dev

dev::LovyanGfx haspTft;
#endif`

End of story. my touch works now!

elik745i, 2023-10-09:
Heres link to video: https://youtube.com/shorts/ETgz38GRDS4?feature=share

elik745i, 2023-10-12:
Don't know how to do that, here are files which needs to be added/replaced.
[Downloads.zip](https://github.com/HASwitchPlate/openHASP/files/12878556/Downloads.zip)

closed 2023-10-12 by elik745i

tinloaf, 2024-05-28:
This should probably not be closed yet, right? Support for the displays is not yet merged into openHASP?

fvanroie, 2024-05-28:
Please create a Pull Request to incorporate the changes into the project.