HASwitchPlate/openHASP #572: Custom MQTT names and alternate topic options

cceaood, 2023-11-10:
The current naming scheme, while perfectly functional is not necessarily the most user-friendly/obvious naming to work with and could, imho be improved for some use cases with some alternate "modes" of operation.

The first thing that comes to mind would be the ability to assign an arbitrary name to an object to be presented to mqtt as well as the option to use the page as part of the topic.

for example  hasp/tv_remote/state/p1b70 could become hasp/tv_remote/state/p1vol_up or hasp/tv_remote/state/p1/vol_up

The second thought on an alternate way to handle this would be to have a single topic for all events and include the id of the object inside the payload (so for example in node red id only need to monitor a single topic).

So we could have hasp/tv_remote/events and each payload would simply have an additional field with the id (and possibly type or object) that generated the event.

MiggyMan, 2023-11-10:
And in spite of logging out and logging in under my own account github still managed to raise this under my work account :D

smcgann99, 2023-11-12:
You can monitor all events with single topic - hasp/plate/state/#
You can also use the tag property to include whatever info you want along with the event -
"tag": {"topic":"someDevice","payload":"WhatToDo"},

cceaood, 2023-11-15:
I'll write something to map the events.

closed 2023-11-15 by cceaood